valadaa48/retroroller #29: usb wifi tplink It doesn't work

snock, 2020-05-19:
dmesg 

```
 [796135.578253] usb 1-3: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
 [796135.578257] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 [796135.578260] usb 1-3: Product: 802.11n NIC
 [796135.578262] usb 1-3: Manufacturer: Realtek
 [796135.578264] usb 1-3: SerialNumber: 00E04C0001
 
```

lsusb

`Bus 001 Device 023: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter`

valadaa48, 2020-05-21:
I'll add this driver for the final image release. Thanks for reporting.

valadaa48, 2020-05-21:
linux-odroid-go2-4.4.189-6 should include that driver. Can you please install from the testing repo and report back?

valadaa48, 2020-05-23:
snockToday at 5:30 AM
helllo
The usb wifi tplink already works for me, THANKS !!

closed 2020-05-23 by valadaa48