valadaa48/retroroller #42: emue-emulationstation doesn't restart through menu

npaladin2000, 2020-05-27:
##Selecting restart EmulationStation in menu doesn't work. Hangs at black screen.

shoesio, 2020-05-27:
Doesn't hang strictly speaking - just closes and leaves a black screen. Launches fine with ssh command

valadaa48, 2020-05-28:
Please update and re-test when ES if your autoboot.

Restart should work if you launch ES through bootup (systemd). Actually quit will also restart it if it's your autobooter.

shoesio, 2020-05-28:
Not behaving as described in previous post for me. 
When I select "restart Emulationstation" from the Emuelec Emulationstation quit menu, it still just exits to a black screen, even with Emulationstation set as autolaunch (I used retro launcher to update this)

shoesio, 2020-05-28:
Selecting "Quit Emulationstation" from emuelec emulation station menu also just closes emulation station, resulting in a black screen

adventureFAN, 2020-05-28:
Reboot/Shutdown works
Can confirm: Restart EmulationStation (for refresh games) dosn't

npaladin2000, 2020-05-28:
Yeah, restart still not working

M00DY-DEV, 2020-05-28:
I can reproduce this.
The only way I can get out is if I ssh: restart

valadaa48, 2020-05-28:
OK this should now be fixed after update + reboot. Please test again.

closed 2020-05-28 by valadaa48