valadaa48/retroroller #43: emuelec-emulationstation spash screen

npaladin2000, 2020-05-27:
Splash screen seems to be hard-coded, need to update with RetroRoller art.

valadaa48, 2020-05-27:
can you find the file that needs change and PR?

npaladin2000, 2020-05-27:
I tried changing the files, it seems to have no effect.

valadaa48, 2020-05-27:
does strace help?

npaladin2000, 2020-05-27:
I'll try that when I get a minute

npaladin2000, 2020-05-27:
There's no reference to any svg files when I run the trace. 

odroid@goadvance:emulationstation$ strace emulationstation
execve("/usr/bin/emulationstation", ["emulationstation"], 0x7fc85a4460 /* 20 vars */) = 0
brk(NULL)                               = 0x5578647000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=87486, ...}) = 0
mmap(NULL, 87486, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f9322b000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\350>\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1795200, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9326d000
mmap(NULL, 1493624, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f930be000
mprotect(0x7f93212000, 65536, PROT_NONE) = 0
mmap(0x7f93222000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x154000) = 0x7f93222000
mmap(0x7f93228000, 10872, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f93228000
close(3)                                = 0
mprotect(0x7f93222000, 12288, PROT_READ) = 0
mprotect(0x555dd0a000, 4096, PROT_READ) = 0
mprotect(0x7f93271000, 4096, PROT_READ) = 0
munmap(0x7f9322b000, 87486)             = 0
brk(NULL)                               = 0x5578647000
brk(0x5578668000)                       = 0x5578668000
execve("/usr/local/sbin/bash", ["bash", "/usr/bin/emulationstation"], 0x7fe0168f38 /* 20 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/local/bin/bash", ["bash", "/usr/bin/emulationstation"], 0x7fe0168f38 /* 20 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/bin/bash", ["bash", "/usr/bin/emulationstation"], 0x7fe0168f38 /* 20 vars */) = 0
brk(NULL)                               = 0x5595df7000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=87486, ...}) = 0
mmap(NULL, 87486, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8ef69000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libreadline.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\20Z\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=359232, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8efab000
mmap(NULL, 428352, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8ef00000
mprotect(0x7f8ef50000, 61440, PROT_NONE) = 0
mmap(0x7f8ef5f000, 36864, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4f000) = 0x7f8ef5f000
mmap(0x7f8ef68000, 2368, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f8ef68000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\340\20\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=14288, ...}) = 0
mmap(NULL, 77920, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8eeec000
mprotect(0x7f8eeef000, 61440, PROT_NONE) = 0
mmap(0x7f8eefe000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f8eefe000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\350>\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1795200, ...}) = 0
mmap(NULL, 1493624, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8ed7f000
mprotect(0x7f8eed3000, 65536, PROT_NONE) = 0
mmap(0x7f8eee3000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x154000) = 0x7f8eee3000
mmap(0x7f8eee9000, 10872, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f8eee9000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libncursesw.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\200b\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=470136, ...}) = 0
mmap(NULL, 535640, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8ecfc000
mprotect(0x7f8ed69000, 65536, PROT_NONE) = 0
mmap(0x7f8ed79000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6d000) = 0x7f8ed79000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8efa9000
mprotect(0x7f8eee3000, 12288, PROT_READ) = 0
mprotect(0x7f8ed79000, 20480, PROT_READ) = 0
mprotect(0x7f8eefe000, 4096, PROT_READ) = 0
mprotect(0x7f8ef5f000, 12288, PROT_READ) = 0
[... 75 lines cut ...]
read(3, "#!/usr/bin/env bash\n\ncd /opt/emu"..., 80) = 65
lseek(3, 0, SEEK_SET)                   = 0
prlimit64(0, RLIMIT_NOFILE, NULL, {rlim_cur=1024, rlim_max=512*1024}) = 0
fcntl(255, F_GETFD)                     = -1 EBADF (Bad file descriptor)
dup3(3, 255, 0)                         = 255
close(3)                                = 0
fcntl(255, F_SETFD, FD_CLOEXEC)         = 0
fcntl(255, F_GETFL)                     = 0x20000 (flags O_RDONLY|O_LARGEFILE)
fstat(255, {st_mode=S_IFREG|0755, st_size=65, ...}) = 0
lseek(255, 0, SEEK_CUR)                 = 0
[... cut ...]
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f8efa90e0) = 12529
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x5589e363b8, sa_mask=[], sa_flags=0}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
wait4(-1, librga:RGA_GET_VERSION:4.00,4.000000
ctx=0x559b9acd00,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
rk-debug mali-ver=rsx-p0-1.2

tlayne, 2020-05-27:
I had no success either when changing various splash logos in the file system. It appears to be baked in. Have we reached out to Shanti?

valadaa48, 2020-05-28:
The file is logo.png in the resources directory. Make a PR when we have one.

M00DY-DEV, 2020-05-28:
Try this. dimensions are 480 x 320.
I tried to keep it as small as possible. If the quality is bad then I can do another one. LMK.

[image]

M00DY-DEV, 2020-05-28:
Try this one:

[image]

valadaa48, 2020-05-28:
looks great! thanks

M00DY-DEV, 2020-05-28:
with the car:

[image]

shoesio, 2020-05-28:
Looking Fly :)

closed 2020-05-28 by valadaa48