valadaa48/retroroller #45: emuelec-emulationstation games settings not working

npaladin2000, 2020-05-27:
As currently configured the Games settings (including controllers settings, per-system and per-game overrides) do not work. As these options are all available in RetroArch manually, we should consider if it's worth the effort to fix them, or simply disable the menu entries.

adventureFAN, 2020-05-27:
And it is not possible to change the core. At least this setting have to work.
Hope the other settings will be also realized. It is one of the best things of the EmuELEC/batocera ES.

npaladin2000, 2020-05-27:
That might be hard to do. The way they have to do that for EmuELEC is have a fixed list of cores to select from and making sure they're all pre-installed and available. If we can do that then I can fix es_systems.cfg to accept the core as a parameter in the launch statement for systems that offer a selection (not all of them will though).

adventureFAN, 2020-05-27:
Why not some core alternatives that are built in by default?
Like Snes9x and Snes9x2010 or FCEUmm and Nestopia or Gambatte and SameBoy?

npaladin2000, 2020-06-05:
Determined to be unnecessary, functionality is in RetroArch.

closed 2020-06-05 by npaladin2000