valadaa48/retroroller #49: easier access to the bios/roms folder

adventureFAN, 2020-05-27:
it is possible to point the bios/roms folder to the odroid folder (roms/bios/ or /home/odroid/bios) for easier handling?

Because "/home/odroid/" is the default "working" folder (like for saves) right?

npaladin2000, 2020-05-27:
/home/odroid (the "odroid" share" is the default working directory for config files. We can consider setting up some shortcuts in the root of that for things like themes, config files, bios files, etc.

npaladin2000, 2020-05-27:
In fact, this should already be in place PR #32

valadaa48, 2020-05-28:
you need to run sudo systemctl start firstboot2.sh to recreate the links

adventureFAN, 2020-05-28:
Dosn't work for me:

odroid@goadvance:~$ sudo systemctl start firstboot2.sh
Failed to start firstboot2.sh.service: Unit firstboot2.sh.service not found.

valadaa48, 2020-05-28:
sudo systemctl start firstboot2 sorry

adventureFAN, 2020-05-29:
okey links to themes and bios folder are now in the odroid folder.
but not the roms folder and I can't access the roms folder anymore? =(
and the link to themes folder also dosn't work. only bios folder working.

npaladin2000, 2020-05-29:
can you SSH in and type "ls -l /" for me? That'll show us the permissions on the roms folder. We might also need "ls -l /roms"

adventureFAN, 2020-05-29:
odroid@goadvance:~$ ls -l /roms
ls: cannot access '/roms/atari2600': Permission denied
ls: cannot access '/roms/gba': Permission denied
ls: cannot access '/roms/neogeo': Permission denied
ls: cannot access '/roms/snes_msu1': Permission denied
ls: cannot access '/roms/cps3': Permission denied
ls: cannot access '/roms/genesis': Permission denied
ls: cannot access '/roms/naomi': Permission denied
ls: cannot access '/roms/mame': Permission denied
ls: cannot access '/roms/snes': Permission denied
ls: cannot access '/roms/gbc': Permission denied
ls: cannot access '/roms/cps2': Permission denied
ls: cannot access '/roms/psp': Permission denied
ls: cannot access '/roms/sgb': Permission denied
ls: cannot access '/roms/scummvm': Permission denied
ls: cannot access '/roms/atari5200': Permission denied
ls: cannot access '/roms/vectrex': Permission denied
ls: cannot access '/roms/atarilynx': Permission denied
ls: cannot access '/roms/megadrive': Permission denied
ls: cannot access '/roms/arcade': Permission denied
ls: cannot access '/roms/pcengine': Permission denied
ls: cannot access '/roms/virtualboy': Permission denied
ls: cannot access '/roms/atomiswave': Permission denied
ls: cannot access '/roms/msx': Permission denied
ls: cannot access '/roms/atari800': Permission denied
ls: cannot access '/roms/nes': Permission denied
ls: cannot access '/roms/ngp': Permission denied
ls: cannot access '/roms/mastersystem': Permission denied
ls: cannot access '/roms/sega32x': Permission denied
ls: cannot access '/roms/nds': Permission denied
ls: cannot access '/roms/cps': Permission denied
ls: cannot access '/roms/sg-1000': Permission denied
ls: cannot access '/roms/cps1': Permission denied
ls: cannot access '/roms/tg16cd': Permission denied
ls: cannot access '/roms/dreamcast': Permission denied
ls: cannot access '/roms/saturn': Permission denied
ls: cannot access '/roms/c64': Permission denied
ls: cannot access '/roms/pcenginecd': Permission denied
ls: cannot access '/roms/wonderswan': Permission denied
ls: cannot access '/roms/n64': Permission denied
ls: cannot access '/roms/gamegear': Permission denied
ls: cannot access '/roms/coleco': Permission denied
ls: cannot access '/roms/intellivision': Permission denied
ls: cannot access '/roms/supergrafx': Permission denied
ls: cannot access '/roms/wonderswancolor': Permission denied
ls: cannot access '/roms/pce': Permission denied
ls: cannot access '/roms/atari7800': Permission denied
ls: cannot access '/roms/fds': Permission denied
ls: cannot access '/roms/msx2': Permission denied
ls: cannot access '/roms/mame2003': Permission denied
ls: cannot access '/roms/dos': Permission denied
ls: cannot access '/roms/segacd': Permission denied
ls: cannot access '/roms/sh': Permission denied
ls: cannot access '/roms/psx': Permission denied
ls: cannot access '/roms/ngpc': Permission denied
ls: cannot access '/roms/gb': Permission denied
ls: cannot access '/roms/sufami': Permission denied
ls: cannot access '/roms/tg16': Permission denied
total 0
d????????? ? ? ? ?            ? arcade
d????????? ? ? ? ?            ? atari2600
d????????? ? ? ? ?            ? atari5200
d????????? ? ? ? ?            ? atari7800
d????????? ? ? ? ?            ? atari800
d????????? ? ? ? ?            ? atarilynx
d????????? ? ? ? ?            ? atomiswave
d????????? ? ? ? ?            ? c64
d????????? ? ? ? ?            ? coleco
d????????? ? ? ? ?            ? cps
d????????? ? ? ? ?            ? cps1
d????????? ? ? ? ?            ? cps2
d????????? ? ? ? ?            ? cps3
d????????? ? ? ? ?            ? dos
d????????? ? ? ? ?            ? dreamcast
d????????? ? ? ? ?            ? fds
d????????? ? ? ? ?            ? gamegear
d????????? ? ? ? ?            ? gb
d????????? ? ? ? ?            ? gba
d????????? ? ? ? ?            ? gbc
d????????? ? ? ? ?            ? genesis
d????????? ? ? ? ?            ? intellivision
d????????? ? ? ? ?            ? mame
d????????? ? ? ? ?            ? mame2003
d????????? ? ? ? ?            ? mastersystem
d????????? ? ? ? ?            ? megadrive
d????????? ? ? ? ?            ? msx
d????????? ? ? ? ?            ? msx2
d????????? ? ? ? ?            ? n64
d????????? ? ? ? ?            ? naomi
d????????? ? ? ? ?            ? nds
d????????? ? ? ? ?            ? neogeo
d????????? ? ? ? ?            ? nes
d????????? ? ? ? ?            ? ngp
d????????? ? ? ? ?            ? ngpc
d????????? ? ? ? ?            ? pce
d????????? ? ? ? ?            ? pcengine
d????????? ? ? ? ?            ? pcenginecd
d????????? ? ? ? ?            ? psp
d????????? ? ? ? ?            ? psx
d????????? ? ? ? ?            ? saturn
d????????? ? ? ? ?            ? scummvm
d????????? ? ? ? ?            ? sega32x
d????????? ? ? ? ?            ? segacd
d????????? ? ? ? ?            ? sg-1000
d????????? ? ? ? ?            ? sgb
d????????? ? ? ? ?            ? sh
d????????? ? ? ? ?            ? snes
d????????? ? ? ? ?            ? snes_msu1
d????????? ? ? ? ?            ? sufami
d????????? ? ? ? ?            ? supergrafx
d????????? ? ? ? ?            ? tg16
d????????? ? ? ? ?            ? tg16cd
d????????? ? ? ? ?            ? vectrex
d????????? ? ? ? ?            ? virtualboy
d????????? ? ? ? ?            ? wonderswan
d????????? ? ? ? ?            ? wonderswancolor

npaladin2000, 2020-05-29:
And what about the first command?

adventureFAN, 2020-05-29:
Can't also update anymore ;_;

`
odroid@goadvance:~$ ls -l /
total 76
lrwxrwxrwx   1 root   root       7 May 23 03:47 bin -> usr/bin
drwxr-xr-x   4 root   root   16384 Jan  1  1970 boot
drwxr-xr-x  13 root   root    3160 May 29 14:07 dev
-rw-r--r--   1 root   root      16 May 29 02:58 ee_arch
drwxr-xr-x  52 root   root    4096 May 29 14:07 etc
drwxr-xr-x   3 root   root    4096 Apr 27 02:29 home
lrwxrwxrwx   1 root   root       7 May 23 03:47 lib -> usr/lib
drwx------   2 root   root   16384 Apr 27 02:24 lost+found
drwxr-xr-x   2 root   root    4096 Nov 19  2019 mnt
drwxr-xr-x   5 root   root    4096 May 27 16:52 opt
dr-xr-xr-x 148 root   root       0 Jan  1  1970 proc
drw-r--r--  59 odroid odroid  4096 May 29 06:31 roms
drwxr-x---   3 root   root    4096 May 25 23:15 root
drwxr-xr-x  18 root   root     500 May 29 14:07 run
lrwxrwxrwx   1 root   root       7 May 23 03:47 sbin -> usr/bin
drwxr-xr-x   4 root   root    4096 Mar  3 14:26 srv
drwxr-xr-x   3 odroid odroid  4096 May 28 10:30 storage
dr-xr-xr-x  14 root   root       0 May 29 14:07 sys
drwxrwxrwt   9 root   root     200 May 29 14:07 tmp
drwxr-xr-x   9 root   root    4096 May 29 06:37 usr
drwxr-xr-x  12 root   root    4096 May 29 06:38 var
`

npaladin2000, 2020-05-29:
Ok, you're missing the exec bit on the /roms folder. Type this:

"sudo chmod -R 755 /roms" and that should fix your ROM access. I'll double-check the script.

npaladin2000, 2020-05-29:
Ok, yep, looks like firstboot was using 644 on that tree, needs to be 755, otherwise you can't browse/change into the folder. I've corrected it, next tiem Val does a build it'll be included.

adventureFAN, 2020-05-29:
Okey thx =)
Now I have access to the rom folder but the rom folder link still dosn't show up in the odroid folder.
The link to the themes folder is still not accessible.

npaladin2000, 2020-05-29:
The roms folder is actually a seperate share, you should have two shares on this IP, the odroid folder and the roms folder. You might need to reboot for it to pick up the roms folder now that it's fixed.  

As far as the themes folder, send me the output of this please:
"ls -la /home/odroid"

adventureFAN, 2020-05-29:
odroid@goadvance:~$ ls -la /home/odroid
total 44
drwx------ 6 odroid odroid 4096 May 29 06:31 .
drwxr-xr-x 3 root   root   4096 Apr 27 02:29 ..
-rw------- 1 odroid odroid  220 May 29 14:29 .bash_history
-rw-r--r-- 1 odroid odroid   21 Feb 17 00:52 .bash_logout
-rw-r--r-- 1 odroid odroid   57 Feb 17 00:52 .bash_profile
-rw-r--r-- 1 odroid odroid  139 May  9 16:01 .bashrc
drwx------ 3 odroid odroid 4096 May 27 16:28 .cache
drwxr-xr-x 6 odroid odroid 4096 May 28 16:16 .config
drwxr-xr-x 8 odroid odroid 4096 May 29 14:30 .emulationstation
-rw------- 1 odroid odroid  885 May 27 16:32 .viminfo
lrwxrwxrwx 1 root   root     35 May 29 06:31 bios -> /home/odroid/.config/retroarch/bios
drwxr-xr-x 2 odroid odroid 4096 May 28 16:37 saves
lrwxrwxrwx 1 root   root     38 May 29 06:31 themes -> /home/odroid/.emulationstation/themes/

npaladin2000, 2020-05-29:
and what happenes when you double-click "themes" in odroid?  according to that the link is there and it's pointing to the right place.

adventureFAN, 2020-05-31:
It is corrected.

closed 2020-05-31 by adventureFAN