valadaa48/retroroller #58: Autosave / load causing issues on Flycast RetroRun launched 32bit version (testing channel)

shoesio, 2020-05-27:
Autosave and load seems to be active on the this version of Flycast. Works ok on some games not on others. Jet Grind Radio seems ok with it. Unreal Tournament just loads to black screen with audio playing. Initial launch of Unreal worked ok with video

shoesio, 2020-05-27:
As you can't enter the Flycast menus on this version no way to fix it for the end user

npaladin2000, 2020-05-27:
RetroRun is launching the 64 bit version (it complains about the 32 bit version, errors out). I'll look into disabling autosave, but that's a serious functionality limit. Which EmulationStation are you using?

shoesio, 2020-05-27:
Just the preinstalled test one currently

npaladin2000, 2020-05-27:
Ok...I switched to the emuelec-testing one, let me see if I can repoduce the issue on that one.

npaladin2000, 2020-05-28:
I haven't been able to reproduce this yet, can you switch to emuelec-emulationstation and try it through there?

shoesio, 2020-05-28:
Same thing happens. Where abouts are save states saved for Flycast? They are not in the roms/dreamcast folder... If I can find the save state will delete and try again

npaladin2000, 2020-05-28:
They should actually be in the home directory

Bryan Pizzuti
npaladin2000@gmail.com

valadaa48, 2020-05-31:
This is a dup of #23 
Can someone help out the flycast dev and test?

npaladin2000, 2020-06-01:
We're moving to 32 bit flycast_xtreme

closed 2020-06-01 by npaladin2000