valadaa48/retroroller #59: Flycast Xtreme analogue controller mapped as digital

shoesio, 2020-05-27:
Can't control in game on Jet Set Radio, so assuming analogue controller is mapped as digital. Can control in game menus with analogue controller ok.

valadaa48, 2020-05-27:
retrorun or retroarch?

shoesio, 2020-05-27:
Launched via Emualtion Station, So I gather it's retrorun

valadaa48, 2020-05-28:
you should maybe report this on the retrorun github

shoesio, 2020-05-28:
Not sure retrorun really has a lot to do with control settings? Will see if I can resolve

shoesio, 2020-05-28:
Just to confirm tried this in Unreal tournament and yes the analog stick is mapped as digital. Shoulder buttons also fail to work which is another common complaint I've seen on the HK forums for batocera - will see if I can work out how they resolved.

valadaa48, 2020-06-05:
I can take retrorun32 from ubuntu20 which should support this?

npaladin2000, 2020-06-05:
I want able to get the retrorun32 from Ubuntu 20 to start. We might need updated 32 bit libraries

npaladin2000, 2020-06-05:
@shoesio  can you try adding a -n (like this) to the command-line in es_systems.cfg please? -n is the analog stick flag

```
	<system>
		<name>dreamcast</name>
		<fullname>Sega Dreamcast</fullname>
		<path>/roms/dreamcast/</path>
		<extension>.gdi .GDI .cdi .CDI</extension>
		<command>performance on; retrorun32 -n -s ~ -d ~/.config/retroarch32/cores ~/.config/retroarch32/cores/flycast_xtreme_libretro.so %ROM%; performance off</command>
		<platform>dreamcast</platform>
		<theme>dreamcast</theme>
	</system>
```

valadaa48, 2020-06-05:
I was able to do it, I can push an update if someone is willing to re-test all 32bit apps

npaladin2000, 2020-06-05:
> I was able to do it, I can push an update if someone is willing to re-test all 32bit apps

Let's see if that -n flag fixes the issue, much less testing that way if so.

npaladin2000, 2020-06-05:
I can test the 32 bit apps

npaladin2000, 2020-06-06:
32 bit apps tested fine, and it seems to alleviate the issue. @shoesio can you update from the testing repo and confirm?

shoesio, 2020-06-06:
So new install, switched to testing repo and updated.
Launching dreamcast title (jet set radio) analog control doesn't seem to work in game. (Slow performance in general.
I've switched out to look in es_systems.cfg and the launcher section for dreamcast doesn't seem to be the 32 bit version - it reads: 
"        <system>
                <name>dreamcast</name>
                <fullname>Sega Dreamcast</fullname>
                <path>/roms/dreamcast/</path>
                <extension>.gdi .GDI .cdi .CDI</extension>
                <command>performance on; retrorun -s ~ -d ~/.config/retroarch/c>
                <platform>dreamcast</platform>
                <theme>dreamcast</theme>
        </system>
" 

Will change manually to look like example above and re-test.

shoesio, 2020-06-06:
OK swapped the config line to the same as the example and JSR runs at good speed (yay!). Analogue stick mapped correctly (yay!) Digital pad mapped correctly (yay!) Shoulder buttons seem to be mapped to L2 and R2 (Not so yay for OGA 1 owners).

npaladin2000, 2020-06-06:
Ok I updated the es systems file to match that. Issue is closed.

closed 2020-06-06 by npaladin2000