valadaa48/retroroller #62: Repeating input on Emulation Station

shoesio, 2020-05-27:
Every now and again the horizontal systems menu goes crazy and starts scrolling on it's own

valadaa48, 2020-05-28:
I can't reproduce this

npaladin2000, 2020-05-28:
Nor am i

Sent from mobile device

M00DY-DEV, 2020-05-28:
Haven't seen this on mine either.

tlayne, 2020-05-28:
I've had this exact issue.

valadaa48, 2020-05-31:
@shoesio is it still happening? If so, can you ssh in and run `evtest` and choose the joypad and see if you're getting spammed without touching controls? Maybe your joystick calibration is off. Please report back.

shoesio, 2020-05-31:
Can't reproduce myself now.  I Ran EVtest and the sticks seems to behave themselves pretty well. I notice now that analogue stick doesn't control menus. Is that a change?  I'd be tempted to close this one out if @tlayne can confirm not effecting hi either now.

tlayne, 2020-05-31:
I switched to EE ES and it’s not an issue there. Feel free to close.

closed 2020-05-31 by valadaa48