valadaa48/retroroller #67: emuelec-es Settings/StartAtBoot

npaladin2000, 2020-05-28:
Menu currently shows EmulationStation and RetroArch as options, and probably neither work. This is redundant versus the EmulationStation/RetroLauncher chooser in retrolauncer and can probably be removed unless we want to adapt it to the RR/ES function.

valadaa48, 2020-05-28:
I fixed this so you can now enable ES or RL at boot

adventureFAN, 2020-05-28:
Can't select this option becaus the "Emuelec Settings" are disabled?
I can go in the menu anymore.

valadaa48, 2020-05-28:
this is fixed, please try again

npaladin2000, 2020-05-28:
I can get at the menu but it doesn't seem to be having an effect, setting for RetroLauncher and rebooting sends me back into ES again.

valadaa48, 2020-05-29:
this works, you need to select it then hit back or B to exit the menu then it takes effect. This is a ES thing.

closed 2020-05-29 by valadaa48