valadaa48/retroroller #73: RA performance mode from RL

npaladin2000, 2020-05-28:
When launching and then exiting RetroArch from RetroLauncher and then exiting, system remains in max performance mode rather than switching to normal/interactive.

valadaa48, 2020-05-28:
I wasn't aware retroarch via retrolauncher sets performance mode. can you check the retrolauncher.toml file to verify?

npaladin2000, 2020-05-28:
It's just the bare command:

[[apps]]
name = "RetroArch"
cmd = "retroarch"
[[apps]]
name = "RetroArch 32-bit"
cmd = "retroarch32"
[[apps]]
name = ""
[[apps]]
name = "Emulation Station"
cmd = "emulationstation"
[[apps]]
name = "PPSSPP"
cmd = "performance on; ppsspp; performance off"

Bryan Pizzuti
npaladin2000@gmail.com

valadaa48, 2020-05-28:
so this is expected, there's nothing to change perf mode so it will stay the same

closed 2020-05-29 by valadaa48