valadaa48/retroroller #74: emuelec-emulationstation only able to launch RetroArch in performance mode

npaladin2000, 2020-05-28:
If performance mode is not enabled in the command or manually activated beforehand then emulationstation quits back to the ROM selection screen.

npaladin2000, 2020-05-28:
WHen attempting to start with performance mode off the following appears on console:

odroid@goadvance:gb$ /usr/bin/retroarch -L ~/.config/retroarch/cores/gambatte_libretro.so Donkey\ Kong\ \(SGB\ Enhanced\).gb
librga:RGA_GET_VERSION:4.00,4.000000
ctx=0x5576a54850,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
MBC1 ROM+RAM+BATTERY loaded.
rambanks: 1
rombanks: 32
could not find encoder!
Segmentation fault (core dumped)
odroid@goadvance:gb$

valadaa48, 2020-05-28:
it works for me with perf mode off

npaladin2000, 2020-05-28:
This is happening on v1.0 hardware, maybe not v1.1 though.

valadaa48, 2020-05-28:
@slaminger  can you check this?

npaladin2000, 2020-05-28:
This appears to be something left over from the 1.8.8 update. I'm going to remove the entire config/retroarch directory so that I can recreate it with a fresh install from non-testing

adventureFAN, 2020-05-28:
Then I have the same issue right? Can it be fixed with an update?

npaladin2000, 2020-05-28:
Not sure yet, I need to see if this works. If it does, fixing this is a manual process.

tlayne, 2020-05-28:
Can confirm, I had to set all systems to perf mode to launch games from EE ES. Also on version 1.0 hardware.

npaladin2000, 2020-05-28:
Slaminger was unable to reproduce this issue, but it turns out he's on the old legacy EmulationStation, so this particular issue is specific to the emuelec-emulationstation package.

valadaa48, 2020-05-28:
Someone should check if this affects Shanti as well. Alternatively, we could force performance mode in the ES .service file and start script?

adventureFAN, 2020-05-28:
In EmuELEC is an option for turning Performance Mode on and off for each system/game.
You can do that in the Game Settings menu.

Perhaps it has something to do with that?

npaladin2000, 2020-05-28:
Maybe. I think we disabled that option but maybe there's some leftover code
relating to it.

Bryan Pizzuti
npaladin2000@gmail.com

valadaa48, 2020-05-29:
This definitely doesn't happen in v11

slaminger, 2020-05-29:
I was able to run nes and GB on model 1 using emuelec es

valadaa48, 2020-06-01:
Can you guys with this issue confirm it's still the case? If so, can you confirm if it happens with the regular ES?

npaladin2000, 2020-06-01:
Still the case on the EE ES.  I don't have the regular one installed
anymore to check.

Sent from mobile device

valadaa48, 2020-06-04:
heard some reports that it's not happening anymore? No v11 users have had this issue. Can someone confirm with a v10?

tlayne, 2020-06-04:
I can test it after my card finishes backing up. About 2 hours from now I should be able to confirm or deny.

slaminger, 2020-06-04:
I can confirm regular es works fine on both. Suggest taking the few
features we want and putting them in simple es vs fixing ee es

tlayne, 2020-06-05:
Tested with EE ES and systems launch as expected now. This issue can be closed.

npaladin2000, 2020-06-05:
THis has been fixed, unknown what particular thing fixed it.

closed 2020-06-05 by npaladin2000