valadaa48/retroroller #79: emuelec-emulationstation Entries that have no Function

adventureFAN, 2020-05-29:
Here are the menu entries that have (in my eyes) no function:

EmuELEC Settings
 - Video Mode
 - Always show boot video (?)

Games Settings 
 - The whole menu (very sadly)

Sound Settings
- Frontend Music (?)

Updates & Downloads
 - The whole menu

System Settingy
- Power Save Mode (?)
- Video Output
- Audio Output
- Overclock
- "Storage" Section (no entry)
- Developer
  - Use RetroArch RGUI Menu
  - Redetect Games Lang/Region
  - unsure about some other entries

valadaa48, 2020-05-29:
should we remove them?

npaladin2000, 2020-05-29:
I won't use them.

adventureFAN, 2020-05-29:
Also Game Settings? *cry*

valadaa48, 2020-05-29:
we can leave that in as a reminder that it has to be fixed. But I can change the label to (not working) ro something

valadaa48, 2020-05-31:
OK I removed most of these. Also sound controls are working now.

closed 2020-05-31 by valadaa48