valadaa48/retroroller #81: Completely exit RetroLauncher when entering Emulation Station

DrBuns3n, 2020-05-29:
I start my OGA using RetroLauncher (RL), but I want minimum overhead for retrorun N64, so when entering Emulation Station (ES) it would be preferable to close RL. I experimented with entering ES from RL then executing  sudo systemctl stop retrolauncher from SSH. As I expected given comments in Discord dialogue about Linux, ES went straight to a black screen. However, sudo systemctl restart emulationstation brought back ES and I tested if RL was also running by executing  sudo systemctl stop retrolauncher again didn't get a black screen. So why can't a script run this sequence to exit properly out of RL when going into ES?

valadaa48, 2020-05-29:
OK I understand now. I don't think this would be easy especially if you plan on going back to RL. I think you may want to wait until RL has file browser mode and skip ES entirely.

DrBuns3n, 2020-05-29:
If RL ends up having a file browser then this would definitely solve the overhead concerns. However executing the same process in reverse allowed me to open RL without ES running in the background too.

valadaa48, 2020-05-29:
you can ssh to the machine and run htop while you are in n64 and verify precisely how many resources RL and ES are taking. Can you paste them here?

DrBuns3n, 2020-05-29:
[image]

valadaa48, 2020-05-29:
you'll notice that both oga-events and RL are using 0% cpu and minimal RAM. ES also seems to be using 0% CPU but a bit more RAM (which is not good for mame) but should be OK here. I don't consider this to be a huge issue as it stands now.

closed 2020-05-30 by valadaa48