valadaa48/retroroller #84: Drastic (strangle frame refresh/drawing)

petervirtual, 2020-05-30:
When running drastic there is a strange look to the frames which almost looks like the you can see frames being drawn left to right (horizontally), it makes games less smooth visually. This issue doesn't seem to exist on emuelec so I have no doubt there is some way to make it draw smoother for a better experience.

valadaa48, 2020-05-30:
@slaminger any ideas?

slaminger, 2020-05-30:
> @slaminger any ideas?

Its sdl2 related I think. Ties in with the menu not refreshing properly.

valadaa48, 2020-05-31:
it's probably the same libsdl2. Maybe check if they have different settings.

tlayne, 2020-05-31:
I’m glad I’m not the only one who experienced this. I didn’t really know how to describe it other than the frames aren’t being drawn to the screen correctly.

npaladin2000, 2020-06-05:
Slaminger pushed an update which fixed this.

petervirtual, 2020-06-06:
was the fix in that latest image?

npaladin2000, 2020-06-07:
Re-opening, issue is occurring again.

npaladin2000, 2020-06-07:
We have switched to Drastic64 to test thig. apparently SDL12 is still showing this issue. Will try SDL10.

npaladin2000, 2020-06-07:
I've switched the installer script to pull Drastic64. If we want to change the URL I'll need to update it again. I also uploaded the v11 configs to the configs directory.  The following is a result of a couple of our tests. 

Ok, our default libSDL2 12 file functions, but only refreshes the menu once every three keypresses. @pppgamer can you confirm if you're still seeing your screen refresh issue with that?

I grabbed libSDL2 10 from Crash's Ubuntu19 image and tried it, it doesn't function at all, crashes right out. 

@valadaa48 is it possible to get a native Arch version of libSDL2-2.o.so.0.10.0 to verify if it's the library itself and not just something goofy with the Ubuntu version trying to work on Arch?

pppgamer, 2020-06-07:
c0a5c02ed83b983a3efad0bca7414102  /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.10.0
66245ed17173f780d1b59897d352a6cc  /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.12.0

both having the screen refresh issue when I test. I am surprised as previously on the non 64bit version 10 would freeze. How sure are we that /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0  path is leveraged by the new 64bit binary?

pppgamer, 2020-06-07:
Do I have the right version f the libs above?

npaladin2000, 2020-06-07:
The 64 bit binary shouldn't be using that at all, it should be using /usr/lib.

npaladin2000, 2020-06-07:
Switched back to 32 bit drastric and released lib32-compat-odroidgo2-1.5.0-1 to fix. Please test.

npaladin2000, 2020-06-07:
Tested OK for me.

npaladin2000, 2020-06-09:
@pppgamer can you make sure the testing repo is enabled for you and it sitting above the regular repo in pacman.conf?

pppgamer, 2020-06-09:
yep. it is, and the package in the testing repo fixed the screen issues

npaladin2000, 2020-06-09:
Closing as fixed.

closed 2020-06-09 by npaladin2000