valadaa48/retroroller #85: General Dreamcast performance

npaladin2000, 2020-05-31:
This issue is to cover discussion/documentation/fixing of the issues running Dreamcast and related (NAOMI/AtomisWave) games on RetroRoller

shoesio, 2020-05-31:
[image]

shoesio, 2020-05-31:
Currently emulation station launches 64bit flycast libretto core using retrorun. Not the fastest option, but sadly Flycast 32bit will not launch correctly from Emulation Station.

npaladin2000, 2020-05-31:
We need to get the 32 bit RetroRun working so we can test that.

npaladin2000, 2020-05-31:
FYI the flycast_xtreme_libretro.so core works great when run maually within RetroArch.

npaladin2000, 2020-05-31:
We will need libevdev.so.2 present to run retroarch32, @valadaa48 please compile and push out when you have a chance

npaladin2000, 2020-06-01:
Ok got my hands on the missing library and tried it. Got the following when tracing it out:

ioctl(12, DRM_IOCTL_GET_CAP, 0xffec24e0) = 0
ioctl(12, DRM_IOCTL_SET_VERSION, 0xffec2494) = -1 EACCES (Permission denied)
write(1, "eglGetPlatformDisplayEXT failed.", 32eglGetPlatformDisplayEXT failed.) = 32
write(1, "\n", 1
)                       = 1
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x38} ---
gettimeofday({tv_sec=1069478512518481, tv_usec=2330128220286306643}, NULL) = 0
write(1, "[err] 00:01:249 libretro/common."..., 112[err] 00:01:249 libretro/common.cpp:373 E[COMMON]: SIGSEGV @ f71f2632 ... 0x38 -> was not in vram (dyna code 0)
) = 112
write(1, "[inf] Fatal error : segfault\n in"..., 84[inf] Fatal error : segfault
 in signal_handler -> core/libretro/common.cpp : 374

) = 84
gettimeofday({tv_sec=1075830769149265, tv_usec=4995127810847819076}, NULL) = 0
write(1, "[err] 00:01:250 libretro/libretr"..., 66[err] 00:01:250 libretro/libretro.cpp:3084 E[COMMON]: DEBUGBREAK!
) = 66
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xf1f58068} ---
gettimeofday({tv_sec=1079069174490449, tv_usec=2330128220286306643}, NULL) = 0
write(1, "[err] 00:01:251 libretro/common."..., 118[err] 00:01:251 libretro/common.cpp:373 E[COMMON]: SIGSEGV @ f1f58068 ... 0xf1f58068 -> was not in vram (dyna code 0)
) = 118
write(1, "[inf] Fatal error : segfault\n in"..., 84[inf] Fatal error : segfault
 in signal_handler -> core/libretro/common.cpp : 374

) = 84
gettimeofday({tv_sec=1083759278777681, tv_usec=4995127810847819076}, NULL) = 0
write(1, "[err] 00:01:252 libretro/libretr"..., 66[err] 00:01:252 libretro/libretro.cpp:3084 E[COMMON]: DEBUGBREAK!
) = 66
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xf1f58068} ---
+++ killed by SIGILL (core dumped) +++
Illegal instruction (core dumped)

npaladin2000, 2020-06-01:
The big difference here is I think retrorun64 was compiled for RetroRoller. I grabbed the retrorun32 binary from stock. We may need it recompiled for our platform.

npaladin2000, 2020-06-02:
We are now testing dreamcast performance using flycast_xtreme_libretro and retrorun32. Seems to be improved.

npaladin2000, 2020-06-05:
We've settled on using retrorun32 and Flycast_xtreme_libretro

closed 2020-06-05 by npaladin2000