valadaa48/retroroller #87: Restart Emulation Station quits back to RS2

DrBuns3n, 2020-05-31:
In the EmuElec version of Emulation Station, the Quit/Restart Emulation Station menu option quits back to RetroLauncher2 instead of restarting emulation station.

valadaa48, 2020-05-31:
This is by design until someone ports a wrapper script to emuelec-es. Restart worked by quitting then touching a file in /tmp then the wrapper script would reload it. EE currently is set to always restart in systemd so if it's your default launcher then quit/restart will restart back into ES. If you use RL2 then EE will quit back to RL2. I think this is a minor annoyance but documenting here if someone can write and test a wrapper script.

valadaa48, 2020-06-06:
this is fixed in the latest version

closed 2020-06-06 by valadaa48