valadaa48/retroroller #93: Minimal theme SNES-CD art

npaladin2000, 2020-06-05:
snes-cd.png needs to be renamed to snesmsu1.png in order to work properly on RetroRoller.

tlayne, 2020-06-05:
I just pushed this change to my repo. I am going to add a few other requested images before bugging Valadaa to repackage.

Can you take a look at this though @npaladin2000 ? I think the theme for these should not be the same otherwise we can't have 2 different logos for these systems.

```
	<system>
		<name>atomiswave</name>
		<fullname>Sega Atomiswave</fullname>
		<path>/roms/atomiswave/</path>
		<extension>.ist .IST .zip .ZIP .bin .BIN</extension>
		<command>performance on; retrorun -d ~/.config/retroarch32/cores ~/.config/retroarch/cores/flycast_libretro.so %ROM%; performance off</command>
		<platform>arcade</platform>
		<theme>sega</theme>
	</system>
	<system>
		<name>naomi</name>
		<fullname>Sega Naomi</fullname>
		<path>/roms/naomi/</path>
		<extension>.ist .IST .zip .ZIP .bin .BIN .dat .DAT</extension>
		<command>performance on; retrorun -d ~/.config/retroarch32/cores ~/.config/retroarch32/cores/flycast_libretro.so %ROM%; performance off</command>
		<platform>arcade</platform>
		<theme>sega</theme>
	</system>
```

npaladin2000, 2020-06-05:
There usually aren't two themes, but here absolutely can be.  Can you coordinate with M00DY to make sure they get added the same way in both of your themes?

tlayne, 2020-06-05:
I pinged him. Hopefully we can come to a consensus.

tlayne, 2020-06-05:
M00D is good with changing the config to 

```
<theme>atomiswave</theme>

<theme>naomi</theme>
```
@npaladin2000 can you update your configs and close this issue?

npaladin2000, 2020-06-05:
Done

closed 2020-06-05 by npaladin2000

M00DY-DEV, 2020-06-06:
@npaladin2000 I added naomi and atomiswave to the theme:

https://github.com/M00DY-DEV/OGA-EmulationStationThemes/tree/master/RetroRama