valadaa48/retroroller #99: Drastic 100% broken

npaladin2000, 2020-06-06:
Not loading at all. Needs to be fixed before we can release.

petervirtual, 2020-06-06:
Here are some notes

c0a5c02ed83b983a3efad0bca7414102  /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.10.0
66245ed17173f780d1b59897d352a6cc  /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.12.0
1270d1644b7b18964bc7eeb14e74e622  /usr/lib/arm-linux-gnueabihf/libevdev.so.2
1270d1644b7b18964bc7eeb14e74e622  /usr/lib/arm-linux-gnueabihf/libevdev.so.2.3.0

When launching drastic with libSDL2-2.0.so.0.12.0 from RL, the menu and screen bugs exist
When launching drastic with libSDL2-2.0.so.0.10.0 from RL, then it freezes

When launching drastic with libSDL2-2.0.so.0.12.0 from ES, drastic doesn't load roms for me and I can't use arrows in the menu to pick one so it's hard to say if the menu or screen bug exists
When launching drastic with libSDL2-2.0.so.0.10.0 from ES, then it freezes (consistent with RL behaviour)

npaladin2000, 2020-06-07:
Fixed

closed 2020-06-07 by npaladin2000

valadaa48, 2020-06-07:
libSDL 0.9.0 appears to work. The controls are atrocious in this emu so I can't really help with much else.