valadaa48/retroroller #101: Theme art update: RetroArch to System

npaladin2000, 2020-06-07:
Currently it themes have an ES entry called RetroArch, but it has since expanded to include several system functions. Can we update the art to reflect this? @tlayne

tlayne, 2020-06-07:
Yes this is just the image that is mapped to system scripts. My image just says Retro. If you want to change the config theme name to something else let me know and I can change the file name on my side.

tlayne, 2020-06-07:
Even better if someone could get me a white png with clear background that says RetroRoller I could use that.

npaladin2000, 2020-06-07:
I just want to change the image, maybe one of the ones that new guy in the channel posted a few days ago might work? THey had a black background so that would be easy enough to convert to transparent.

tlayne, 2020-06-07:
Sure! Send it my way.

npaladin2000, 2020-06-08:
Actually I found your white png too

[image]
[image]

npaladin2000, 2020-06-08:
Oh what's m00dy's id on here? We should tag him too

tlayne, 2020-06-08:
@M00DY-DEV

s4ilence, 2020-06-08:
[image]

here is the png for that image transparent background.. if you want to use it

npaladin2000, 2020-06-08:
While I do like the new one, I'm thinking we should probably be consistent and use the current one. I'll leave the new one up there in case anyone wants to build a new theme around it though.

tlayne, 2020-06-08:
Since my theme isn’t the default I feel like it would be okay for me to use the unofficial one. What do you guys think?

npaladin2000, 2020-06-08:
I like both logos. But since we're offering both as installable packages, they're both to some degree "official" and I guess should use the official art.  But that's my opinion, I'll leave it up to you guys and @valadaa48

M00DY-DEV, 2020-06-08:
Hey @tlayne is this what you're using?

https://raw.githubusercontent.com/tlayne/es-theme-minimal/master/Minimal/art/png/retroarch.png

M00DY-DEV, 2020-06-08:
@tlayne also, do you want to change your config to point to a new file or do you still want to use retroarch.png? I recommend rr.png or retroroller.png.

What do you guys think of this to replace Retroarch, it's sized to the image you're using for your Minimal theme:

[image]

npaladin2000, 2020-06-08:
I like it.  Everyone, to see it make sure you click on that blank looking white space :)

tlayne, 2020-06-08:
Downloaded it. @npaladin2000 you are the one that manages the config so let us know if you want to change the config name under theme and we will match our PNG's accordingly.

npaladin2000, 2020-06-08:
Let's just use the same PNG and theme names and replace the art. Fewer changes that way, and less chances for me to make a typo :)

tlayne, 2020-06-08:
sounds good you can close if M00D is ready.

npaladin2000, 2020-06-08:
@M00DY-DEV let us know when you're set and I'll close this out.

M00DY-DEV, 2020-06-08:
I just updated the theme with these images. If you guys are good with then feel free to close it out.

[image]
[image]

npaladin2000, 2020-06-08:
Oh wow, that is VERY sharp.

closed 2020-06-08 by npaladin2000