valadaa48/retroroller #110: RetroArch cores missing from image

npaladin2000, 2020-06-14:
The following probably need to be included in the image at the start:

Picodrive (SMS, genesis/megadrive, gg, segacd, 32x
mGBA (GBA, SuperGB)

valadaa48, 2020-06-14:
It's a lot of work creating images and I can't keep doing this indefinitely. I think there needs to be clear post-install documentation on how to online update packages + cores.

npaladin2000, 2020-06-14:
I agree, but if we're going to have a core preconfigured for use it should probably already be present in the image rather than having to retrieve it. Probably just note this down for the next image you make, should be no need to re do the image just for this.

valadaa48, 2020-06-15:
Is it only picodrive that is missing?

mikepurvis, 2020-11-12:
I would even support going the other direction, and trimming it right down to the bare-bones. I ended up uninstalling the cores that shipped on the RRVL image in favour of ones installed using the online updater, as none of the ones it came with actually worked for me.

closed 2020-12-13 by valadaa48