valadaa48/retroroller #112: Stability fix for rtl8812au and rtl8811au chipset

christianhaitian, 2020-06-14:
For those that have a Odroid Wifi Module 5a or dongle with the rtl8811au or rtl8812au chipset and are experiencing unstable wifi on retroroller, I recommending putting the file in the included zip in the /etc/modprobe.d folder then rebooting the device.  It contains a setting to disable the operating system power management for this chipset which improves stability.
[rtl8812au.zip](https://github.com/valadaa48/retroroller/files/4776936/rtl8812au.zip)

valadaa48, 2020-06-14:
Thanks for this, do you think you could add this to the wiki somewhere where it will be more permanent? I also have this module and can test it out.

christianhaitian, 2020-06-14:
> Thanks for this, do you think you could add this to the wiki somewhere where it will be more permanent? I also have this module and can test it out.

Done.

valadaa48, 2020-06-14:
Thanks! This should be very helpful to others.

closed 2020-06-14 by valadaa48