valadaa48/retroroller #114: Dedicated snes9x2010 launcher?

asoderq, 2020-06-14:
Hi,

could we have a dedicated Snes9x2010 launcher in ES? Of course I can configure it myself but I would like to use upstream config.
There are a few games that doesn't run full speed on the latest Snes9x.
I would simply call the system snes9x2010 with correspondingly named folder in /roms/ 

Also in the released image there seems to be a redundant folder under roms called snes_msu1, while snesmsu1 seems to be used by ES.

npaladin2000, 2020-06-14:
The Sufami entry is already set up to use Snes9X2010 if you'd like to use it. In fact we've had so few requests for that system we might end up converting to an Snes9x201 entry eventually.

asoderq, 2020-06-15:
I saw that yes. If the reason that Sufami uses Snes9x2010 is performance, then yes I would support the idea of giving it a more generic name as other enhancement chip games also don't run full speed on latest snes9x. But also in general I don't see a reason of having more entries in ES, as they don't show up if you don't use them. But you would also don't want it to get out of hands. E.g. snes and sfc both exist which is ok. But I probably wouldn't want folders depending region for all systems.

npaladin2000, 2020-06-15:
We could probably rename it to SNES-Enhanced, that would cover SuperFX, SA-1, and the Sufami setup I guess. We'd need new theme are though @tlayne

asoderq, 2020-06-15:
Are we 100% that we wont find a game that doesn't run full speed on snes9x2010? Or that some mode 7 games might also not run full speed on the latest core, which would trigger us to also have snes9x2002/5. In this case I am for calling it the name of the core.

tlayne, 2020-06-15:
@npaladin2000 just let me know what’s decided and I will accommodate.

npaladin2000, 2020-06-15:
Ok let's convert the Sufami entry then. Let M00d know?  

@asoderq we don't have the 2002 or 2005 cores available anyway

asoderq, 2020-06-15:
You can download them on Retroarch32.
Would you be willing to support 2002 if I find games that doesn't run full speed on the 2010 core?
In that case I would test the ones I think are the most likely to not run full-speed.

One of the nice things with the oga is that it can emulate snes full speed, albeit sometimes you have to compromise the accuracy.

npaladin2000, 2020-06-15:
No, we're trying to reduce the dependency on 32 bit applications and libraries. We can look into those cores for 64 bit, if we can find a reason to include them, but there would have to be several games where it gets the best performance, and no other core that provides acceptable performance.

asoderq, 2020-06-15:
I have played the beginning of all FX games and many SA-1 (including some hacks) and many other games with other enhancement chips on the snes9x2010 and didn't find any major performance problems.

npaladin2000, 2020-06-15:
@tlayne I don't know which is easier for you on this, if you just want to retain the art name or change it. Let me know what you and M00d decide and I'll update the configs.

tlayne, 2020-06-15:
Can you summarize the proposed changes? Are we adding more entries or consolidating entries?

npaladin2000, 2020-06-15:
We're going to change the "sufami" entry into an "SNES Enhanced" entry to run SNES stuff with SA-1 or SuperFX chips, or the twin-cartridge Sufami setup.

tlayne, 2020-06-15:
Okay, @npaladin2000 a new entry for super charged SNES games should be easy enough. Just let me know what the theme label in your config is and I will whip one up.

npaladin2000, 2020-06-15:
Let's call it snes-e

tlayne, 2020-06-15:
Okay I added it, but you will need to build pkg again so it will show up in updates.

Commit is 9f89645

npaladin2000, 2020-06-15:
Ok I'll go approve it and build the package later

closed 2020-06-15 by npaladin2000