valadaa48/retroroller #120: Wifi controls inaccessible with Retroarch as default launcher

Cwiiis, 2020-06-18:
It's really nice to have the device boot into Retroarch, but doing so means you lose the ability to control whether wifi is on/off and to connecto to new/different networks.

It'd be good if either there was a hotkey combination to bring up the wifi app, or alternatively, to bring up RetroLauncher to access everything (though I guess RetroLauncher isn't really designed to be quit? Really, it's just the wifi I want, personally)

valadaa48, 2020-06-18:
I have a branch of retroarch that supports wifi config, I just haven't gotten around to PR'ing it. I think it will eventually get added. But you can launch retrolauncher through spring no?

Cwiiis, 2020-06-18:
> I have a branch of retroarch that supports wifi config, I just haven't gotten around to PR'ing it. I think it will eventually get added. But you can launch retrolauncher through spring no?

I've not noticed any menu items for it, where are they located? Having wifi config support in retroarch would be great though...

valadaa48, 2020-06-18:
it wasn't merged it but I'll update here when it is

valadaa48, 2020-06-24:
You should also be able to launch the NetworkSetup.sh (or whatever it's called) from /roms/sh, can you confirm if that works?

valadaa48, 2020-09-01:
RRVL retroarch 1.9.0-3 adds built-in Wifi configurability.

valadaa48, 2020-09-01:
https://github.com/libretro/RetroArch/pull/11267

closed 2020-09-01 by valadaa48