valadaa48/retroroller #122: spring mupen retroarch 64 core crashes on rom load

Cwiiis, 2020-06-18:
Trying to load any N64 rom with the spring mupen core (which I assume is meant to launch mupen, like the other spring cores) just causes retroarch to quit (I assume segfault, but I don't know how to check just yet). Running mupen from the console works fine.

Cwiiis, 2020-06-18:
Dug a bit more into this - mupen64plus works fine as root, but not as user odroid. When run as odroid, I get this output:

```
librga:RGA_GET_VERSION:4.00,4.000000
ctx=0x157b7eb0,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
 __  __                         __   _  _   ____  _
|  \/  |_   _ _ __   ___ _ __  / /_ | || | |  _ \| |_   _ ___
| |\/| | | | | '_ \ / _ \ '_ \| '_ \| || |_| |_) | | | | / __|
| |  | | |_| | |_) |  __/ | | | (_) |__   _|  __/| | |_| \__ \
|_|  |_|\__,_| .__/ \___|_| |_|\___/   |_| |_|   |_|\__,_|___/
             |_|         https://mupen64plus.org/
Mupen64Plus Console User-Interface Version 2.5.9

UI-Console: attached to core library 'Mupen64Plus Core' version 2.5.9
UI-Console:             Includes support for Dynamic Recompiler.
Core: Using full mem base
Core: Goodname: F-ZERO X (E) [!]
Core: Name: F-ZERO X
Core: MD5: EE79A8FE287B5DCAEA584439363342FC
Core: CRC: 776646F6 06B9AC2B
Core: Imagetype: .z64 (native)
Core: Rom size: 16777216 bytes (or 16 Mb or 128 Megabits)
Core: Version: 1449
Core: Manufacturer: Nintendo
Core: Country: Europe (0x50)
UI-Console Status: Cheat codes disabled.
UI-Console Error: dlopen('./reicast_libretro.so') failed: ./reicast_libretro.so: undefined symbol: REMOVED_OPS
UI-Console: using Video plugin: <dummy>
UI-Console Error: dlopen('./reicast_libretro.so') failed: ./reicast_libretro.so: undefined symbol: REMOVED_OPS
UI-Console: using Audio plugin: <dummy>
UI-Console Error: dlopen('./reicast_libretro.so') failed: ./reicast_libretro.so: undefined symbol: REMOVED_OPS
UI-Console: using Input plugin: <dummy>
UI-Console Error: dlopen('./reicast_libretro.so') failed: ./reicast_libretro.so: undefined symbol: REMOVED_OPS
UI-Console: using RSP plugin: <dummy>
Core Warning: No video plugin attached.  There will be no video output.
Core Warning: No RSP plugin attached.  The video output will be corrupted.
Core Warning: No audio plugin attached.  There will be no sound output.
Core Warning: No input plugin attached.  You won't be able to control the game.
Core: Using video capture backend: dummy
Core: Game controller 0 (Standard controller) has nothing plugged in
Core: Game controller 1 (Standard controller) has nothing plugged in
Core: Game controller 2 (Standard controller) has nothing plugged in
Core: Game controller 3 (Standard controller) has nothing plugged in
Core: Using CIC type X106
Core: Initializing 4 RDRAM modules for a total of 8 MB
Core: Starting R4300 emulator: Dynamic Recompiler
Core: Init new dynarec
```

If I run it as root:
```
librga:RGA_GET_VERSION:4.00,4.000000
ctx=0x15f0beb0,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
 __  __                         __   _  _   ____  _
|  \/  |_   _ _ __   ___ _ __  / /_ | || | |  _ \| |_   _ ___
| |\/| | | | | '_ \ / _ \ '_ \| '_ \| || |_| |_) | | | | / __|
| |  | | |_| | |_) |  __/ | | | (_) |__   _|  __/| | |_| \__ \
|_|  |_|\__,_| .__/ \___|_| |_|\___/   |_| |_|   |_|\__,_|___/
             |_|         https://mupen64plus.org/
Mupen64Plus Console User-Interface Version 2.5.9

UI-Console: attached to core library 'Mupen64Plus Core' version 2.5.9
UI-Console:             Includes support for Dynamic Recompiler.
Core: Using full mem base
Core: Goodname: F-ZERO X (E) [!]
Core: Name: F-ZERO X
Core: MD5: EE79A8FE287B5DCAEA584439363342FC
Core: CRC: 776646F6 06B9AC2B
[... 19 lines cut ...]
Video: Disabled SSE processing.
Video: ROM (CRC f64666772bacb906-50) not found in INI file
Video: Enabled hacks for game: 'F-ZERO X'
Video: Initializing OpenGL Device Context.
rk-debug mali-ver=rsx-p0-1.2
EGL: major=1, minor=4
EGL: Vendor=ARM
EGL: Version=1.4 Bifrost-"r6p0-01rel0"
EGL: ClientAPIs=OpenGL_ES
EGL: Extensions=EGL_KHR_partial_update EGL_KHR_image_pixmap EGL_EXT_image_dma_buf_import EGL_KHR_config_attribs EGL_KHR_image EGL_KHR_image_base EGL_KHR_fence_sync EGL_KHR_wait_sync EGL_KHR_gl_colorspace EGL_KHR_get_all_proc_addresses EGL_IMG_context_priority EGL_ARM_pixmap_multisample_discard EGL_ARM_implicit_external_sync EGL_KHR_gl_texture_2D_image EGL_KHR_gl_renderbuffer_image EGL_KHR_create_context EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image EGL_EXT_create_context_robustness
EGL: ClientExtensions=EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_x11 EGL_EXT_platform_x11 EGL_KHR_platform_gbm

Video: Using OpenGL: Mali-TDVX - OpenGL ES 3.2 v1.r6p0-01rel0.abe21775750b0ba1ad10e2a1a5cf9985 : ARM
Audio: Using resampler speex
Audio: Initializing SDL audio subsystem...
Core: Initializing 4 RDRAM modules for a total of 8 MB
Core: Starting R4300 emulator: Dynamic Recompiler
Core: Init new dynarec
Audio: Initializing SDL audio subsystem...
Core Status: Stopping emulation.
Core: R4300 emulator finished.
Core Status: Rom closed.
```

Note, I played some and then closed the emulator normally in that last log (thus the closing messages at the end). Copying the config over from the root config fixed the issue of running it as user odroid, but doesn't fix the issue of launching it from retroarch :( Worth noting, if I run it from the libretro cores directory, it will always load the dummy video/audio/input backends, even if I move reicast_libretro.so out of the way - not sure why that is... I assume it must have a hard-coded library path of the current directory and something in there is conflicting.

Cwiiis, 2020-06-18:
My theory now is that mupen64plus looks in the directory you launch it from and either then doesn't fall back to the default plugins directory, or hits some kind of failure state that prevents it from launching its own plugins (I assume the dummy plugins are built in)

Cwiiis, 2020-06-18:
Yes, executing mupen64plus in /usr/lib produces similar bad results (I don't recommend doing this...) - so either mupen64plus needs to be patched to not do this, or spring launcher needs to make sure to switch directory before executing mupen64plus.

Cwiiis, 2020-06-18:
Looks like plugin searching is done in mupen64-plus-ui-console. I'm just reading through the code, possibly the config file could be setup so this doesn't happen...

valadaa48, 2020-06-18:
can you try to change directories in the spring config?

Cwiiis, 2020-06-18:
Just testing a patch, will submit a PR if it works.

Cwiiis, 2020-06-18:
> can you try to change directories in the spring config?

I tried this first, but couldn't get it to work - I think it's worth patching out loading random libraries from mupen anyway, that's awful default behaviour :)

Cwiiis, 2020-06-18:
Well, the patch didn't work, but this can be worked around by specifying the plugins on the path, so if you change the execution from just `mupen64plus` to `mupen64plus --gfx mupen64plus-video-rice --audio mupen64plus-audio-sdl.so --input mupen64plus-input-sdl.so --rsp mupen64plus-rsp-hle.so`, you can launch it from the retroarch cores directory. Not sure why my patch didn't work, but there you go, easier to work around then fix...

Cwiiis, 2020-06-18:
So fixing this by renaming the mupen64plus binary and replacing it with a script that does the right thing works - but it seems the launcher in retroarch is still broken, it just quits out... But if I run retroarch from the terminal it works... ¯\_(ツ)_/¯

Cwiiis, 2020-06-18:
Finally got it working - there are a number of issues... After replacing mupen64plus with a script that passes the full path of all 4 plugins, for some reason retroarch is only giving the rom name and not its path (perhaps spring launcher needs to be doing something more/different here?) I've hard-coded the directory in my launcher script and now it works as expected :) Of course, it ends up quitting back to your default launcher rather than retroarch after it finishes, but that's a different issue...

I think someone with better knowledge of the system would be better off fixing this in the correct way (by fixing the spring launcher), but this at least works for me and all the information above lets you know how to fix it.

valadaa48, 2020-06-24:
Maybe you can wrap that in a script and we can distribute it for people that want to run from the command line?

valadaa48, 2020-06-26:
Alternatively, you can paste this to the wiki or create a spring launcher .info file for others to use?

Cwiiis, 2020-06-26:
I'm not really happy with my solution - like, why does it only give the rom name and not the full path? If a user had changed their rom directory, my script wouldn't work, so it'd be good to figure at least that out first... Maybe it'd just be a better idea to build the Mupen64Plus core though, rather than relying on launching the external program anyway? I might look into that, certainly it'd be nice to have the RetroArch UI and features when using it.

valadaa48, 2020-06-29:
all I had to do myself was `mupen64plus %s` for spring launcher and it worked. I'll check again

valadaa48, 2020-07-09:
Closing since we have confirmation it's working for others

closed 2020-07-09 by valadaa48