valadaa48/retroroller #123: Hot Key functionality broken after doing System Update

chrismillr, 2020-06-18:
Using F3 + Up/Down, Left/Right, and R/L to set brightness, volume, and governors seems to no longer work after performing a system update this morning.

npaladin2000, 2020-06-18:
It was switched, F2 I think.

chrismillr, 2020-06-18:
Thanks. That fixed the issue. Neither works on the main boot application though. The hotkey is still listed as F3, while F2 launches RA32.

npaladin2000, 2020-06-18:
@valadaa48 I assume this is going to require a config change to retrolauncher.toml?

valadaa48, 2020-06-18:
yes it will

valadaa48, 2020-06-19:
configs are updated

chrismillr, 2020-06-19:
[image]
Its still not showing properly on RetroLauncher after performing a System Update.

nvllsvm, 2020-06-21:
@chrismillr I had to manually update ~/.config/retrolauncher/retrolauncher.toml after the update.

valadaa48, 2020-06-23:
system update doesn't change user configs and until we have a config-synchronization mechanism all config related changes need to be merged/copied manually from the configs dir @npaladin2000 .

closed 2020-06-23 by valadaa48