valadaa48/retroroller #124: Hotkey for killing app(s)

Cwiiis, 2020-06-20:
It'd be great to have a hotkey to kill the currently launched app - I guess this might be hard to ascertain, in which case, a hotkey to just kill all the launchers and emulators that the system ships with perhaps? If you use an unreliable retroarch core and it freezes in an unrecoverable way, the only recourse is to power off the device (unless you have a machine handy nearby and wifi was enabled)

valadaa48, 2020-06-23:
We've thought about this but it's a tricky thing unless we launch all apps through another app so we know which ones to kill (child). If you can think of anything then PRs are considered.

Cwiiis, 2020-06-23:
I wonder if you could look (via something like lsof?) to see what apps have the framebuffer open and kill those? I'll take a quick look next time I get a minute.

tlayne, 2021-02-05:
Hotkey kill is a feature for some RRVL "Springs" so I would consider this resolved. Hotkey kill can be configured separately in Retroarch by the user if preferred.

closed 2021-03-02 by valadaa48