valadaa48/retroroller #125: No screen output after complete battery depletion and recharge

n1zzo, 2020-06-21:
Hi,
thanks for the amazing project, it's great to have a mini arch-based retro gaming console.
I left the odroid go discharge completely, after charging it up again and powering up:
- the uboot splashscreen is shown
- the screen flickers a bit
- no more screen output
- network is functioning, ssh is functioning
- launching `/usr/bin/retrolauncher` from ssh correctly shows the ncurses menu

Do you have any suggestions to fix the issue and avoid its occurence in the future?

valadaa48, 2020-06-23:
You might want to ask on the forums about this. I doubt the issue you're experiencing is OS related but in case of corruption you can try to reimage RR or even another OS to verify?

n1zzo, 2020-06-23:
I can confirm that I flashed retro roller from scratch and the issue was resolved, probably it was just some kind of fs corruption.
I'm going to close this.

Thank you for the assistance!

closed 2020-06-23 by n1zzo