valadaa48/retroroller #126: dosbox-staging from pacman not working

suetlung, 2020-06-23:
libretro dosbox core is too slow.
I am trying to install dosbox-staging from pacman and get it run.

It started but nothing showed up. Do I need to set anything in the conf?

valadaa48, 2020-06-23:
Someone here got it working and has a config. If they didn't post it to the wiki then perhaps you can ask around in the RetroGame Handhelds discord.

valadaa48, 2020-06-29:
pcgamer on discord is the person I think who got it working, not sure if he posted his config

suetlung, 2020-06-30:
Sorry I am not so familiar with Discord. I joined https://discord.gg/p4uRmCd but cannot see OGA channel. Do I need to do something to see it? Sorry for stupid questions.

valadaa48, 2020-12-19:
Closing this in favor of dosbox-staging in RRVL.

closed 2020-12-19 by valadaa48