valadaa48/retroroller #134: volume go back to 100% temperately after resume from sleep

NOirBRight, 2020-06-29:
no matter what volume level you set.
If you go to sleep and resume, the volume will go back to 100% for about 1 second then set to you previous level.
both in archlinux and rrvl.

valadaa48, 2020-06-29:
do you know if this happens in other images?

valadaa48, 2020-06-30:
The reason I ask is because I thought joy fixed this issue in the kernel. I fixed this in rrvl. Please test and if it works then I'll apply to RR.

NOirBRight, 2020-06-30:
no, the volume issue is still there.
this issue was also in rr2.0 early build, but fixed later.

NOirBRight, 2020-06-30:
[image]
it fixed in rr2.0 after I pointed it.

NOirBRight, 2020-06-30:
the volume will be 100% after resume for about 1 second then go back to what you set in RRVL.

NOirBRight, 2020-06-30:
1.0.3_2 seems fixed this.

closed 2020-06-30 by valadaa48