valadaa48/retroroller #141: Make a Ports Spring core

NeonLightning, 2020-07-01:
title says it all. if someone wants to make a spring for running sh scripts and pr it that'd be great.

valadaa48, 2020-07-01:
Actually I don't really understand this. Can you explain what this means?

NeonLightning, 2020-07-01:
oh as was said on discord. me and noir were thinking there could be a spring to load any sh instead of making springs for each port. and just have users have a folder of port scripts themselves. it'll allow for more than just ports potentially.

valadaa48, 2020-07-01:
Can you provide examples? Right now you can make a spring_sh that loads any sh script you browse. I guess I don't really understand the ask here.

NeonLightning, 2020-07-01:
that's what i'm saying. is just including one for the enduser preinstalled would be good.

valadaa48, 2020-07-02:
I made a spring_shell in rr-base 1.1.1, test that out with your .sh to launch your port

NeonLightning, 2020-07-02:
i mentioned it on discord but it worked

closed 2020-07-02 by NeonLightning