valadaa48/retroroller #144: Retroarch Ozone menu text too small to read

shoesio, 2020-07-02:
Ozone menu text is very small and difficult to read. UI scaling is functional for Ozone now, but has a max limit of 1.5. Icons and battery indicator are also difficult to make out at any size.

valadaa48, 2020-07-02:
I think @npaladin2000 has a workaround explained in the wiki

npaladin2000, 2020-07-02:
Yeah, it would be nice if that could be in the image by default though (not make Ozone the default, just pre-remove the ttf files).

shoesio, 2020-07-02:
The fonts are loaded when you update assets. Deleting the ttf font doesn't really sort the issue out though - just makes it look a bit more low tech (and admittedly readable). Is it possible to make the UI scaling work up to a larger number, for those of us who like a nice smooth font?

valadaa48, 2020-07-03:
I can look into patching retroarch, will need some time for this though.

shoesio, 2020-07-03:
FIXED

closed 2020-07-03 by shoesio