valadaa48/retroroller #148: RGA and Bilinear scaling broken in Retroarch 32bit (Void)

shoesio, 2020-07-03:
Turning on either RGA scaling or Bilinear filter crashes Retroarch32

valadaa48, 2020-07-04:
From or chat we assume this is due a combination of spring and retroarch restarting after changing these settings? Not sure there is an easy fix.

valadaa48, 2020-07-09:
Can we try to narrow down the scope of this issue. Are we specifically talking about retroarch32 itself crashing when switching these modes or do you merely think it's "crashing" because it restarts back to retroarch64? If it's the latter then, I'm afraid there's nothing to do at this point other than pray we get rid of 32bit cores. Also, if it's again the latter, the safest thing might be to boot into retroarch32 directly and set a per-core option so you don't need to change this stuff again.

valadaa48, 2020-07-15:
Feel free to re-open if you still think this is an issue that can be reasonably addressed.

closed 2020-07-15 by valadaa48