valadaa48/retroroller #149: Drastic menu unresponsive (Void)

shoesio, 2020-07-03:
Latest update of Drastic introduces bug that makes menu in Drastic not display every button press. Only every third press is updated onscreen, although first and second presses are processed.

To reproduce: Press F1 to enter menu.
Use Pad to move between menu options, only one in three presses are updated onscreen.

valadaa48, 2020-07-03:
I can't reproduce. Can anyone else?

SeongGino, 2020-07-08:
I'm not sure if this counts as 'reproduce' but it's affecting me here too.

Also the framerate is dramatically reduced, even at points where it should be running fine without frameskip.

valadaa48, 2020-07-09:
Can you try changing /usr/bin/drastic to:

```
#!/bin/sh

cd /opt/drastic
LD_PRELOAD=/usr/lib/libSDL2-2.0.so.0.10.0 /drastic "$@"
```

and see if it helps?

SeongGino, 2020-07-09:
That does seem to fix the problem.

valadaa48, 2020-07-09:
OK thanks for confirming. I guess there a few things we can do here:

1) Force 2.0.10 system-wide and get rid of 2.0.12 completely. I think EmuELEC does this.
2) All SDL2 apps that have this issue needs to have a bash script to do the LD_PRELOAD dance.

Any thoughts?

shoesio, 2020-07-09:
Menu issue seems fixed, but showing new issues in emu now. Experienced a crash, slower performance and some kind of flashing / black frame insertion.

valadaa48, 2020-07-09:
does this happen in EE?

shoesio, 2020-07-09:
No, menu reacts fine

SeongGino, 2020-07-09:
> 
> 
> Menu issue seems fixed, but showing new issues in emu now. Experienced a crash, slower performance and some kind of flashing / black frame insertion after frame skip being applied

Can't reproduce, and performance difference is difficult to gauge when the original performance pre-fix felt as if it was layered beneath four degrees of Frameskip.

shoesio, 2020-07-09:
Just testing some more now, performance in game is very variable, moves from silky smooth to half speed with nothing on screen particularly causing it.

shoesio, 2020-07-09:
I think the black frame thing was my battery dying to be honest. Having played a good half hour on it I haven't noticed that again. Performance is still variable in normal play however.

valadaa48, 2020-07-10:
are you running in performance mode?

shoesio, 2020-07-10:
The spring should ensure that right?

valadaa48, 2020-07-15:
try updated and ensure SDL2-2.0.10-go2 is removed and SDL2-go2 is downgraded to 2.0.10. I tested it and can't reproduce now.

SeongGino, 2020-07-15:
Updated, also can't reproduce.

valadaa48, 2020-07-15:
Thanks, let's close this for now as I think we got it.

closed 2020-07-15 by valadaa48