valadaa48/retroroller #152: RRVL: wifi does not work

klundry, 2020-07-08:
I'm using an original Go Advance with an edimax wifi dongle which works fine with arch based retroroller but does not work with RRVL. I have edited the wpa_supplicant.conf with my credentials.

Safarikniv, 2020-07-08:
Have you edited wpa_supplicant.conf on the the boot partition, or directly in /etc?

Boot partition is the correct place.

klundry, 2020-07-08:
I edited the file on the boot partition.

valadaa48, 2020-07-09:
Do you know what kernel you were using in RR? was it the specialized rr kernel in testing or was it the default one?

klundry, 2020-07-09:
The default. I didn't compile a new kernel or install a different one.

valadaa48, 2020-07-09:
It could be related to wpa_supplicant or it could be related to the driver not being in the kernel. Are you able to try the -rr kernel in RR to see if your device is recognized? I think you need to enable the testing-repo for that. If it works, then I suspect it's wpa_supplicant.

valadaa48, 2020-07-09:
kernel -3 adds a boat load of wifi drivers. Can you provide the lusb output of your edimax?

klundry, 2020-07-10:
I get command not found when I try to run lsusb but dmesg gives me this. I'm not sure if that's helpful or not. 

[    6.851335] usbcore: registered new interface driver rtl8192cu

klundry, 2020-07-28:
Wifi is working now after updating via usb ethernet adapter.

valadaa48, 2020-08-31:
RRVL final image will be released so hopefully this won't be an issue.

closed 2020-08-31 by valadaa48