valadaa48/retroroller #155: Is it possible to include ResidualVM in RR or RRVL

er0l, 2020-07-14:
I have already made this question on odroid go forums, but here it is for tracking purpose.

ResidualVM is a cross-platform 3D game interpreter which allows you to play LucasArts' Lua-based 3D adventures: Grim Fandango and Escape from Monkey Island, provided you already have their data files. ResidualVM just replaces the executables shipped with the games, allowing you to play them on systems for which they were never designed!

ResidualVM is using SDL2 and it is very similar to ScummVM, but it is not available as libretro core.

valadaa48, 2020-09-01:
@er0l RRVL has a residualvm package already built. You can install it with `sudo xbps-install -S residualvm`. You may have to force SDL2 or gles2 mode for it to launch. Let me know if it works for you.

valadaa48, 2021-03-02:
We can use opks for this

closed 2021-03-02 by valadaa48