valadaa48/retroroller #156: systemd update of latest RC causes system crash

scheckley, 2020-07-16:
Hi,

Odroid advance go black edition, running latest RC image. System crashes when updating systemd to 245.6-8 (using ssh to execute pacman -Syu).

system unrecoverable.

valadaa48, 2020-07-16:
We've seen this issue before and it seems that when systemd is updated, it loses wifi for a bit. What we found worked was to give it 5-10 mins to update before rebooting if wifi doesn't come back.

RRVL doesn't suffer from this issue if you prefer to give that a try.

scheckley, 2020-07-16:
> We've seen this issue before and it seems that when systemd is updated, it loses wifi for a bit. What we found worked was to give it 5-10 mins to update before rebooting if wifi doesn't come back.
> 
> RRVL doesn't suffer from this issue if you prefer to give that a try.

I'll give RRVL a try, thanks :)

valadaa48, 2020-07-17:
how did you fare with RRVL?

scheckley, 2020-07-17:
Hi,

I only just set it up, but it seems to work ok. OS updated fine over ssh. PSX, Genesis, and SNES emulation work so far using the bundled retro arch.

closed 2020-09-01 by valadaa48