valadaa48/retroroller #164: TIC-80 core for RR and RRVL

klundry, 2020-07-29:
I was able to compile the TIC-80 core easily on RetroRoller but I am getting an error with cmake on RRVL. Can we get the core added to the repo?

valadaa48, 2020-07-30:
@npaladin2000 or safarikniv on discord should be able to help get this core added to the repo. Alternatively you can copy it over manually. Thanks for compiling it.

klundry, 2020-07-30:
Yes, I've copied it to .config/retroarch/cores and its working.

ofuangka, 2020-08-19:
@klundry Can you can share the compiled core so it can be added to the repo? I'd been having issues compiling it myself.

klundry, 2020-08-20:
Sure. How would you like me to share it with you?

valadaa48, 2020-08-31:
Is it possible to post a link to it? We can add it to the retroarch core repo.

klundry, 2020-08-31:
@valadaa48 Forgot you can upload file to github directly in the issues.
[tic80_libretro.zip](https://github.com/valadaa48/retroroller/files/5151564/tic80_libretro.zip)

valadaa48, 2020-09-12:
@klundry I added this core from my own optimized build to the core downloader

closed 2020-09-12 by valadaa48