valadaa48/retroroller #165: How to switch from retroarch to retrolauncher on RRVL?

klundry, 2020-07-29:
I have installed retrolauncher on RRVL and I'm assuming I can disable retroarch launching by default by doing removing the symlink in /var/service but I don't see a retrolauncher service in /etc/sv.

valadaa48, 2020-07-30:
I think for now until it's fixed in the package you can try to create a service: you should be able to copy retroarch from /etc/sv to retroaluncher and edit the command in the file in that directory and try it out. If you get it working please post back here and that will help with integration.

valadaa48, 2020-09-05:
@klundry FYI, this is being worked on!

klundry, 2020-09-08:
Awesome, thanks!

valadaa48, 2020-12-13:
There is a RetroLauncher available in RRVL now. Not sure if you want to use it as I think RetroArch is just as powerful with Spring launchers. I'll close this for now.

closed 2020-12-13 by valadaa48

klundry, 2020-12-14:
It is available but does not work unfortunately. It just boots to a blinking cursor. I personally would like to have it but realize I'm probably in the extreme minority :).