valadaa48/retroroller #166: RRVL - PCSX Spring - alignment fault with CHDs

kaeltis, 2020-08-09:
Hi,

first of all, I love RRVL! It works great with almost everything I throw at it.
Thanks for creating such an awesome distro for our OGAs! 😃

I've just found one issue so far where something refuses to work:

After playing a CUE/BIN PSX ROM via the PCSX ReARMed Spring (which worked perfectly), I decided to try some of my CHD files.
However trying to load a .chd file crashes PCSX with the following error, I tried it with the oga and gl display driver:
```
[  694.352605] rockchip-vop ff460000.vop: [drm:vop_crtc_enable] Update mode to 320x480p60, type: 16
[  694.369451] dw-mipi-dsi ff450000.dsi: final DSI-Link bandwidth: 450 x 1 Mbps
[  695.890256] Unhandled fault: alignment fault (0x92000061) at 0x00000000f08ea13c
```

As far as I understand, the forked version of PCSX ReARMed should be able to handle those CHDs without any problems, so it might be some OGA specific bug?

valadaa48, 2020-08-12:
Thanks for the kind comments and glad you like the distro!

That paste seems like like it's from the kernel. I'm not able to debug this right now but would be curious to know if the pcsx rearmed or retroarch github has any reported issue of loading chds. It could also just have been a bad build. 

Can anyone else here comment if CHDs work for them on RRVL?

BadBrent, 2020-08-12:
CHDs work just fine in RRVL since the core supports the format, alongside ISO, BIN/CUE, PBP, MDF, IMG, CCD, and a few other weird ones that are probably related to running the game off of the actual disc inside of an optical drive connected to the device playing the game.  CHDs and PBPs however are the only format that are able to be compressed and I haven't had the best luck with PBP games in the past.  I'm personally experiencing CHD slowdowns at times in certain games that don't even require much from a PS1 system let alone the emulator core so it's strange figuring out just how much I/O is going into optical drive image files especially if they are compressed.  CHD format definitely gives you the smallest images that you can play directly in RetroArch and I have yet to see a single game that can't be ran in this format on PCSX ReARMed.  

Tangent: I might have to go on a case by case basis and see which version runs better on the OGA, CHD or PBP since I have no plans of storing the uncompressed CD image on my SD card since that's up to 700MB+ per disc.  The entire PS1 NTSC library is 1700 games or so and has an overall size of 680GB but once you compress them all into CHDs you can shave the collection down to just 388GB.  That allows me the ability to store every game ever released for the PS1 on a single 512GB micro SD card.  Now I need to perform a mass conversion of bin/cue to PBP format but not compress the crap out of the games since you lose fidelity of audio as well as texture quality in many games plus some games have issues with loading in certain spots....just like what I'm dealing with when using CHDs right now.

kaeltis, 2020-08-12:
> That paste seems like like it's from the kernel. I'm not able to debug this right now but would be curious to know if the pcsx rearmed or retroarch github has any reported issue of loading chds. It could also just have been a bad build.

Oh yes, forgot to add that, it gets dumped to the console/dmesg when the Core/RetroArch crashes.
I could try attaching gdb to RetroArch and see if that shows anything meaningful, don't have a lot of experience with it though.

>CHDs work just fine in RRVL

Thats weird, I created those CHDs with chdman and they work perfectly fine on my PC with the Windows build of RetroArch with both Beetle PSX HW and PCSX ReARMed, they just crash on RRVL.

valadaa48, 2020-09-01:
@kaeltis is this still an issue for you? Were you able to track what's going on?

kaeltis, 2020-09-01:
@valadaa48 yes, I was now able to test a couple of things using the following command I got from the spring file:
`retroarch32 --verbose -L /usr/arm/opt/libretro/pcsx_rearmed_libretro.so "/roms/psx/Crash Bandicoot (USA).chd"`

Running the command from the odroid user results in the same above kernel fault, this happens with and without the BIOS:
```
odroid@rrvl:[~]: retroarch32 --verbose -L /usr/arm/opt/libretro/pcsx_rearmed_libretro.so "/roms/psx/Crash Bandicoot (USA).chd"
librga:RGA_GET_VERSION:4.00,4.000000
ctx=0xab9d8098,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
[INFO] [Overrides]: Redirecting save file to "/home/odroid/.config/retroarch/saves/Crash Bandicoot (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/home/odroid/.config/retroarch/states/Crash Bandicoot (USA).state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Aug 29 2020
[INFO] Version: 1.9.0
[INFO] =================================================
[ERROR] Couldn't find any wifi driver named "nmcli"
[INFO] Available wifi drivers are:
[INFO]  null
[WARN] Going to default to first wifi driver...
[INFO] [CORE]: Loading dynamic libretro core from: "/usr/arm/opt/libretro/pcsx_rearmed_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/PCSX-ReARMed.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/psx.cfg.
[INFO] [Overrides]: No game-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/Crash Bandicoot (USA).cfg.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Remaps]: remap directory: /home/odroid/.config/retroarch/config/remaps
[INFO] [Overrides]: Redirecting save file to "/home/odroid/.config/retroarch/saves/Crash Bandicoot (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/home/odroid/.config/retroarch/states/Crash Bandicoot (USA).state".
[INFO] [Environ]: GET_MESSAGE_INTERFACE_VERSION.
[libretro INFO] Memcard 2: disabled
[libretro INFO] Starting PCSX-ReARMed
[libretro INFO] Running PCSX Version 1.9 (Jun 28 2020).
[libretro INFO] Init new dynarec
[libretro INFO] testing if we can run recompiled code..
[libretro INFO] test passed.
[libretro INFO] Mapped (RAM/scrp/ROM/LUTs/TC):
[libretro INFO] 80000000/1f800000/1fc00000/08000000/f33ef000
[INFO] [Environ]: SYSTEM_DIRECTORY: "/home/odroid/.config/retroarch/bios".
[libretro INFO] found BIOS file: scph101.bin
[INFO] [Environ]: GET_CAN_DUPE: true
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 6.
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[libretro INFO] RGB565 supported, using it
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[libretro INFO] plugin: plugins/builtin_gpu
[libretro INFO] plugin: plugins/builtin_spu
[libretro INFO] plugin: plugins/builtin_pad
[libretro INFO] plugin: plugins/builtin_pad
[... 127 lines cut ...]
[INFO] Disconnecting device from port 8.
[INFO] Disconnecting device from port 9.
[INFO] Disconnecting device from port 10.
[INFO] Disconnecting device from port 11.
[INFO] Disconnecting device from port 12.
[INFO] Disconnecting device from port 13.
[INFO] Disconnecting device from port 14.
[INFO] Disconnecting device from port 15.
[INFO] Disconnecting device from port 16.
[INFO] [Playlist]: Loading history file: [/root/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/root/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/root/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/root/.config/retroarch/content_favorites.lpl].
[ERROR] failed_to_start_audio_driver
[INFO] SET_GEOMETRY: 640x478, aspect: 1.333.
[INFO] Loading state: "/root/.config/retroarch/states/Crash Bandicoot (USA).state.auto".
[INFO] State size: 4456448 bytes.
[INFO] Saving state: "RAM".
[INFO] State size: 4456448 bytes.
[INFO] State size: 4456448 bytes.
ari64_reset
[INFO] SET_GEOMETRY: 512x240, aspect: 1.333.
```

Could this be some permission issue while accessing the oga display driver?

kaeltis, 2020-09-01:
@valadaa48 I didn't have more time to test it, but I think I just figured it out, that fault / bus error only happens with pcsx rearmed when retro achievements are enabled.

```
odroid@rrvl:[~]: retroarch32 --verbose -L /usr/arm/opt/libretro/pcsx_rearmed_libretro.so "/roms/psx/Crash Bandicoot (USA).chd"
librga:RGA_GET_VERSION:4.00,4.000000
ctx=0xabbbd098,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
[INFO] [Overrides]: Redirecting save file to "/home/odroid/.config/retroarch/saves/Crash Bandicoot (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/home/odroid/.config/retroarch/states/Crash Bandicoot (USA).state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Aug 29 2020
[INFO] Version: 1.9.0
[INFO] =================================================
[ERROR] Couldn't find any wifi driver named "nmcli"
[INFO] Available wifi drivers are:
[INFO]  null
[WARN] Going to default to first wifi driver...
[INFO] [CORE]: Loading dynamic libretro core from: "/usr/arm/opt/libretro/pcsx_rearmed_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/PCSX-ReARMed.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/psx.cfg.
[INFO] [Overrides]: No game-specific overrides found at /home/odroid/.config/retroarch/config/PCSX-ReARMed/Crash Bandicoot (USA).cfg.
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: GET_CORE_OPTIONS_VERSION.
[INFO] [Environ]: GET_LANGUAGE: "0".
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_INTL.
[INFO] [Remaps]: remap directory: /home/odroid/.config/retroarch/config/remaps
[INFO] [Overrides]: Redirecting save file to "/home/odroid/.config/retroarch/saves/Crash Bandicoot (USA).srm".
[INFO] [Overrides]: Redirecting save state to "/home/odroid/.config/retroarch/states/Crash Bandicoot (USA).state".
[INFO] [Environ]: GET_MESSAGE_INTERFACE_VERSION.
[libretro INFO] Memcard 2: disabled
[libretro INFO] Starting PCSX-ReARMed
[libretro INFO] Running PCSX Version 1.9 (Jun 28 2020).
[libretro INFO] Init new dynarec
[libretro INFO] testing if we can run recompiled code..
[libretro INFO] test passed.
[libretro INFO] Mapped (RAM/scrp/ROM/LUTs/TC):
[libretro INFO] 80000000/1f800000/1fc00000/08000000/f345d000
[INFO] [Environ]: SYSTEM_DIRECTORY: "/home/odroid/.config/retroarch/bios".
[libretro INFO] no BIOS files found.
[WARN] [Environ]: SET_MESSAGE_EXT: No PlayStation BIOS file found - add for better compatibility
[INFO] [Environ]: GET_CAN_DUPE: true
[INFO] [Environ]: GET_DISK_CONTROL_INTERFACE_VERSION.
[INFO] [Environ]: SET_DISK_CONTROL_EXT_INTERFACE.
[INFO] [Environ]: GET_RUMBLE_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 6.
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[libretro INFO] RGB565 supported, using it
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[INFO] [Environ]: RETRO_ENVIRONMENT_SET_CORE_OPTIONS_DISPLAY.
[libretro INFO] plugin: plugins/builtin_gpu
[libretro INFO] plugin: plugins/builtin_spu
[libretro INFO] plugin: plugins/builtin_pad
[libretro INFO] plugin: plugins/builtin_pad
[libretro INFO] Plugins loaded.
[libretro INFO] Loaded CD Image: /roms/psx/Crash Bandicoot (USA).chd[+chd].
[libretro INFO] Track 01 (DATA) - Start 00:02:00, Length 59:43:18
selected sound output driver: libretro
[libretro INFO] CD-ROM Label: SCUS-94900
[libretro INFO] CD-ROM ID: SCUS94900
[libretro INFO] CD-ROM EXE Name: SCUS_949.00;1
ari64_reset
[libretro INFO] note: running with HLE BIOS, expect compatibility problems
[libretro INFO] ----------------------------------------------------------
[INFO] [SRAM]: Skipping SRAM load..
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Cheats]: Load game-specific cheatfile: /home/odroid/.config/retroarch/cheats/PCSX-ReARMed/Crash Bandicoot (USA).cht
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Video @ 480x320
[INFO] [Video]: Starting threaded video driver ...
Bus error
```

valadaa48, 2020-09-02:
Bus error looks to be a memory access issue. I remember reading about some RetroAchievements (cheevos) bugs. If you do an RRVL update (check the CHANGELOG), you'll get 1.9.0 and perhaps it's been fixed?

kaeltis, 2020-09-02:
I created that log after updating to the most recent version yesterday, so the issue seems to persist.

valadaa48, 2020-10-25:
We might have figured out the issue here. We were compiling retroarch with LTO and I heard that was causing issues with chds. The new update disables that if you want to try again?

kaeltis, 2020-10-27:
I've tried it and it's still causing an alignment fault as soon as achievements are enabled, without them it still works perfectly.
I've updated to the latest version though, I haven't tried installing a new image.

valadaa48, 2020-12-11:
retroarch-odroidgo2-1.9.0_23 fixes this issue. Can you please update and confirm back?

Thanks

valadaa48, 2020-12-13:
We've had confirmation that this is fixed.

closed 2020-12-13 by valadaa48