valadaa48/retroroller #167: Command to proot into system on x86_64, to update, run installs scripts, etc?

cplr, 2020-08-11:
Hi, I'm trying to use `proot` to `chroot` into the RetroRoller SD filesystem and run the pacman update, run install scripts like to install drastic, etc. The reason for this being, I don't have a working networking device for my 1.0 OGA, and so I'm just trying to use my ubuntu VM to do this.

Please note that the "unable to lock database" error seen below is *not* solvable using the usual suggestion (removing the `db.lck` file - there is no `db.lck` file).

Has anyone successfully used `proot` or `chroot` to do this on an x86_64 system? If so, this would be very handy to have in the wiki.

```
me@ubuntu-minimal-xfce:~$ proot -S /media/me/retroroller-mount-sd-point/ -q qemu-aarch64
root@ubuntu-minimal-xfce:~# pacman -Syu
:: Synchronizing package databases...
error: failed to update core (unable to lock database)
error: failed to update extra (unable to lock database)
error: failed to update community (unable to lock database)
error: failed to update alarm (unable to lock database)
error: failed to update aur (unable to lock database)
error: failed to update retroroller (unable to lock database)
error: failed to synchronize all databases
```

M00DY-DEV, 2020-08-11:
Hey, check the bottom of the wiki:
https://github.com/valadaa48/retroroller/wiki/Updating

cplr, 2020-08-11:
see my 2nd paragraph - there is no db lock file, that's not the solution. this is some error related to trying to update while `proot`-ed. I feel this is possible but I am just not mapping the right directories (the `-S` option automaps directories that it thinks are most appropriate)

valadaa48, 2020-08-12:
I don't think I've ever used proot directly for this purpose before. I generally chroot using qemu-user-static and provided /var, /tmp, /dev are properly mounted with --bind I've never had an issue updating via pacman. 

If there is a verbose flag you could try turning that on to see if it gives you more information. Alternatively, the archlinux community or the wiki/bbs might help?

cplr, 2020-08-12:
Could you give an example `chroot` command? I'm new to this aspect of Linux and thought `proot` would be more useful, but perhaps I'm using the wrong tool for the job. 

Thanks so much for answering and for all your hard work!

valadaa48, 2020-08-12:
```# cd /location/of/new/root
# mount -t proc /proc proc/
# mount --rbind /sys sys/
# mount --rbind /dev dev/

# chroot /location/of/new/root /bin/bash
# source /etc/profile```

something along those lines, but you'll need qemu-user-static and binfmt enabled to be able to run binaries. You can google for your distribution on how to enable that.

valadaa48, 2020-08-30:
@cplr is this still an issue for you? If not can we close?

cplr, 2020-08-30:
Unfortunately I haven't had a chance to test your suggestions, but the comment that this might be useful for the wiki still stands.

I do appreciate your guidance with this matter. I wish there were more hours in the workday.

valadaa48, 2020-12-13:
closing due to inactivity

closed 2020-12-13 by valadaa48