valadaa48/retroroller #172: Adding 7zr and unrar to /usr/bin (feature request)

mariovirgili, 2020-08-12:
Is it possible to add some archiving utilities such as 7zr and unrar? 
It would be a nice help for those willing to extract files from copied .7z or .rar archives directly on SD.

In alternative, it would be better choice if .7z or .zip support can be added to the Dreamcast, Saturn, N64 roms launching.

valadaa48, 2020-08-16:
I don't think they're included in the base image but you should be able to easily install them

mariovirgili, 2020-08-16:
Yeah, if you know how to do it it’s all easy. 😉I discovered how to use pacman to install them, but it would be simple for all if they were included in the base image.

valadaa48, 2020-08-30:
These have been added to rr-base 1.3.1_2 in RRVL.

```
Name    Action    Version           New version            Download size
p7zip   install   -                 16.02_5                1151KB 
unzip   install   -                 6.0_14                 146KB 
rr-base update    1.3.1_1           1.3.1_2                29KB
```

closed 2020-08-30 by valadaa48