valadaa48/retroroller #176: RRVL proposal

mariovirgili, 2020-08-29:
Ok, I'm giving another chance to RRVL! :D Just kidding.
I have a request: before launching you need to edit wpa_supplicant.conf using another pc, windows or linux.
What about only asking for a keyboard to be connected to the usb port and insert SSID and password at first run?
You would only need to run a script like this once:

echo "Please insert WiFi network name (SSID)"; read network; echo "Please insert your password"; read password

printf "\nnetwork={\nssid=\"$network\"\npsk=\"$password\"\n}\n" >>./wpa_supplicant.conf

valadaa48, 2020-08-29:
That's a good idea but I'm not sure how many people have USB keyboards at hand or are willing to go through that rigamarole. The next update of retroarch will include configuring the wifi but using NetworkManager. Hopefully this will mitigate the need for people to edit the file in /boot and especially thwart Windows corrupting their partitions.

mariovirgili, 2020-08-29:
I think that if they have a pc they surely have a keyboard attached that can be used one-time only. :)

shoesio, 2020-08-30:
A lot of laptops and wireless keyboards out there these days @mariovirgili

mariovirgili, 2020-08-30:
You’re right, my fault. I’m a boomer.. ;)

valadaa48, 2020-08-31:
I'm going to close this since the new retroarch for RRVL (soon to be released) has built-in wifi configuration.

closed 2020-08-31 by valadaa48