valadaa48/retroroller #177: RRVL: libGLESv2.so error fix

mariovirgili, 2020-08-29:
Find out why RRVL was not working for me.

This time I had no kernel panic while installing it. 
Straight out of the box, after downloading the cores needed, were working GBA, N64 via spring, PCFX and Saturn.
But it failed to open libretro core for N64 (parallel), dreamcast and naomi, with the following error:

[INFO] [CORE]: Loading dynamic libretro core from: "/home/odroid/.config/retroarch/cores/flycast_libretro.so"
[ERROR] Failed to open libretro core: "/home/odroid/.config/retroarch/cores/flycast_libretro.so"
Error(s): libGLESv2.so: cannot open shared object file: No such file or directory

A quick link solved the problem:

sudo ln -s /usr/lib/mali/libGLESv2.so /usr/lib/libGLESv2.so

valadaa48, 2020-08-29:
I'm confused why libGLESv2.so wasn't picked up from /usr/lib/mali. 

```[odroid@rrvl ~]$ cat /etc/ld.so.conf.d/libmali-bifrost.conf 
/usr/lib/mali```

valadaa48, 2020-08-29:
Loading parallel n64 worked for me without that workaround.

mariovirgili, 2020-08-29:
I just used the image linked in the wiki with a fresh sd card.
And it was the second time that I come up with that issue

valadaa48, 2020-08-29:
Did you do a full update?

mariovirgili, 2020-08-29:
Sincerely I do not remember if I update before or after the fix. I can retry to write image and test it..

valadaa48, 2020-08-29:
I suspect a later package update fixes it, no need to try to reproduce if it's too much trouble.

mariovirgili, 2020-08-29:
It’s not a final working sd, it’s only a test system with few roms inside. I can wipe all and re-copy Bios and roms

valadaa48, 2020-08-30:
Thanks, please let us know what you find. I appreciate your testing.

mariovirgili, 2020-08-30:
Left my comment on discord.

valadaa48, 2020-08-30:
Thanks for testing, sounds like we need a base image.

mariovirgili, 2020-08-30:
Yes, but... what is a base image? A working updated image?

valadaa48, 2020-08-31:
I'm preparing a final RRVL image and confirm that this doesn't have the issue. I assume that once the 20200629 image is updated this issue will go away. I'll close this for now but feel free to re-open if we see this issue again.

closed 2020-08-31 by valadaa48