valadaa48/retroroller #178: Assorted problems in RRVL

DrBuns3n, 2020-08-31:
flycast 64 bit: failed to open libretro core

no spring launcher for retrorun flycast32

mupen64plus-next: failed to open libretro core

spring mupen64plus quits back to RA (is this supposed to be the retrorun spring?)

start+select to quit to RA hotkeys do not work in Stella 2014 and Prosystem 

spring Drastic quits back to RA

mariovirgili, 2020-08-31:
To fix some problems: sudo ln -s /usr/lib/mali/libGLESv2.so /usr/lib/libGLESv2.so

DrBuns3n, 2020-08-31:
[odroid@rrvl ~]$ sudo ln -s /usr/lib/mali/libGLESv2.so /usr/lib/libGLESv2.so
ln: failed to create symbolic link '/usr/lib/libGLESv2.so': File exists

Received the above error trying to apply the fix.

DrBuns3n, 2020-08-31:
After applying sudo ln -s /usr/lib/mali/libGLESv2.so /usr/lib/libGLESv2.so, the following now works:
- flycast 64 bit
- spring mupen64plus

valadaa48, 2020-09-01:
I reproduced this libGLESv2.so issue, it was caused by rockchip's new drivers. I'll be rolling these back soon.

valadaa48, 2020-09-01:
> flycast 64 bit: failed to open libretro core

this is fixed in latest RRVL

> no spring launcher for retrorun flycast32

Please create a separate issue to track adding retrorun

> mupen64plus-next: failed to open libretro core

This should work now

> spring mupen64plus quits back to RA (is this supposed to be the retrorun spring?)

This should launch the standalone mupen64plus and I confirmed it works in the new update

start+select to quit to RA hotkeys do not work in Stella 2014 and Prosystem

> I think F1+F2 are the hotkeys, not start+select

spring Drastic quits back to RA

> This is tested as working in the latest update

closed 2020-09-01 by valadaa48

valadaa48, 2020-09-01:
retrorun and retrorun32 were added as packages in the next update.