valadaa48/retroroller #182: RRVL sleep is unreliable

klundry, 2020-09-04:
When pressing the power button to enter sleep mode it usually will go to sleep as expected. However, it almost never wakes up from sleep successfully. Just doesn't respond to the power button press to wake up and I have to force power off.

valadaa48, 2020-09-04:
The team can't reproduce this at all. Do you have a BE or is it the v1?

klundry, 2020-09-04:
It is a v1 and I have just discovered it appears to be causes by my WiFi dongle. Without it plugged in the sleep is working perfectly so far.

valadaa48, 2020-09-05:
Thanks for narrowing it down. I'm curious if this happens with other images that support sleep? I double checked the code for suspend and resume and it was ported from RR and I don't remember hearing about too many resume issues.

klundry, 2020-09-05:
I'm not sure but I will test that when I get a chance.

valadaa48, 2020-09-06:
can you run `lsmod` and see if your dongle's driver is loaded as a module? If it's the case, then you might be able to rmmod it before sleep then readd it. You can check in /etc/zzz.d/ for these hooks.

valadaa48, 2020-09-11:
We just pushed an update that includes wifi firmware. Please update and reboot to see if it helps?

valadaa48, 2020-12-10:
@klundry is this still an issue for you?

klundry, 2020-12-11:
Yes it is still happening for me with this wifi dongle. I looked it up and its an RTL8188CUS chipset. Here is what I get from lsmod.

> Module                  Size  Used by
> 8192cu                724681  1
> rtl8xxxu               68555  0
> rtl8192cu              95628  0
> rtl_usb                15264  1 rtl8192cu
> rtl8192c_common        60413  1 rtl8192cu
> rtlwifi                96300  3 rtl_usb,rtl8192c_common,rtl8192cu
> dwc2                  140816  0
> vhost_net              12740  0
> vhost                  33425  1 vhost_net
> hci_vhci                5229  0
> tun                    23671  1 vhost_net

valadaa48, 2020-12-13:
I wonder why rtl8xxxu is being loaded. I suppose you can try to unload that module and see if it works. Alternatively, you can try to unload/load it before suspend/wake in /etc/zzz.d (you should find similar treatment of dwc2 there too).

closed 2021-03-02 by valadaa48