valadaa48/retroroller #183: RRVL with Ralink RT5370: No WiFi

deBaer, 2020-09-06:
Hi, I (only) have this USB WiFi dongle (148f:5370) for my OOGA that worked great with the ArchRR, but on RRVL, it doesn't work. The LED in the dongle doesn't turn on, and RetroArch finishes the WiFi scan in a nanosecond, finding no SSIDs. Since I cannot figure out how to get a terminal without WiFi, I cannot send any debug information.

deBaer, 2020-09-06:
Forgot to mention I'm using the latest image, from September 4th. Also, manually entering my WiFi information into network.sh doesn't help.

valadaa48, 2020-09-07:
Sorry for your issues. I think I see what may be causing your issue. The USB driver is compiled as a module whereas your WIFI's driver is compiled as a builtin (statically). I just compiled a new experimental kernel and what I can do is release a test image for you and the other person here that is having issues with their wifi if you'd be willing to reflash and test?

valadaa48, 2020-09-07:
After some further investigation, it looks like the driver is being compiled as a module same as RR. The issue we see here might in fact be because VoidLinux doesn't include a required firmware file. If you still have RR handy, could you post the output of `lsusb -v`, `lsmod`? I suspect it's the rt2800usb driver. If this is listed in lsmod, please also post `modinfo rt2800usb`. Thanks for your help with this!

valadaa48, 2020-09-07:
Alternatively, I found this: https://forums.gentoo.org/viewtopic-t-1044248-start-0.html

Copying /usr/lib/firmware/rt2870.bin to /usr/lib/firmware/rt3070.bin might work.

deBaer, 2020-09-07:
Hi, I was able to hook up a USB hub and an Ethernet adapter, so I have access to a shell. `dmesg` gives me

```[ 2301.316735] usb 1-1.1: reset high-speed USB device number 6 using dwc2
[ 2301.431783] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 5390, rev 0502 detected
[ 2301.461147] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 5370 detected
[ 2301.461808] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[ 2301.462839] usbcore: registered new interface driver rt2800usb
[ 2301.501261] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2301.501573] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[ 2301.502320] rt2800usb 1-1.1:1.0: Direct firmware load for rt2870.bin failed with error -2
[ 2301.502355] ieee80211 phy0: rt2x00lib_request_firmware: Error - Failed to request Firmware
[ 2301.506886] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[ 2301.506976] rt2800usb 1-1.1:1.0: Direct firmware load for rt2870.bin failed with error -2
[ 2301.506998] ieee80211 phy0: rt2x00lib_request_firmware: Error - Failed to request Firmware
[ 2301.508689] usb 1-1.4: new high-speed USB device number 9 using dwc2
```

The interesting part of `lsusb -v`:

``` Bus 001 Device 006: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x148f Ralink Technology, Corp.
  idProduct          0x5370 RT5370 Wireless Adapter
  bcdDevice            1.01
  iManufacturer           1 Ralink
  iProduct                2 802.11 n WLAN
  iSerial                 3 1.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              450mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           7
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              5 1.0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
[... 350 lines cut ...]
alias:          usb:v14B2p3C23d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v14B2p3C12d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v14B2p3C09d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v14B2p3C07d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v14B2p3C06d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p3072d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p3071d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p3070d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p2870d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p2770d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v8516p2070d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1EDAp2310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1EDAp2210d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1EDAp2012d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1482p3C09d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p3072d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p3071d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p3070d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p2770d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p2870d*dc*dsc*dp*ic*isc*ip*in*
depends:        rt2x00usb,rt2x00lib,rt2800lib
intree:         Y
vermagic:       4.4.189_6 SMP preempt mod_unload aarch64
parm:           nohwcrypt:Disable hardware encryption. (bool)
```

deBaer, 2020-09-07:
OK, I just grabbed rt2870.bin from the first Google hit, [http://www.lwfinger.com/firmware/](http://www.lwfinger.com/firmware/), and now WiFi is working. I could have found that out quicker, but not without a shell. Springing into a shell from "load core" doesn't work for me, it just dumps me back into RetroArch.

valadaa48, 2020-09-09:
Thanks for figuring this out. Where did you put rt2870.bin? I'm very curious because RRVL should have this by default: /usr/lib/firmware/rt2870.bin

deBaer, 2020-09-10:
Sadly, it's not. This a newly flashed SD with the most current image:

```
[odroid@rrvl ~]$ ls -l /usr/lib/firmware/
total 1160
-rw-r--r-- 1 root root 201138 Nov 20  2018 ipw2100-1.3-i.fw
-rw-r--r-- 1 root root 196458 Nov 20  2018 ipw2100-1.3-p.fw
-rw-r--r-- 1 root root 209190 Nov 20  2018 ipw2100-1.3.fw
-rw-r--r-- 1 root root 191154 Nov 20  2018 ipw2200-bss.fw
-rw-r--r-- 1 root root 185428 Nov 20  2018 ipw2200-ibss.fw
-rw-r--r-- 1 root root 187836 Nov 20  2018 ipw2200-sniffer.fw
drwxr-xr-x 2 root root   4096 Jun 28 04:21 zd1211
[odroid@rrvl ~]$
```

valadaa48, 2020-09-11:
This is a bug. I added the wifi firmware to the next update and also a new base image to help out others with the same issue. Thanks for finding this!

To test it out, just run update and reboot to ensure your network works.

closed 2020-09-11 by valadaa48