valadaa48/retroroller #184: RRVL rgb10 usb network doesn't work on boot but works after re-plug

Jimmy-Z, 2020-09-07:
As title, this is the corresponding dmesg output
```
[  +0.034499] usb 1-1: new high-speed USB device number 2 using dwc2
[  +0.201926] usb 1-1: New USB device found, idVendor=0b95, idProduct=1790
[  +0.000025] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000012] usb 1-1: Product: AX88179
[  +0.000010] usb 1-1: Manufacturer: ASIX Elec. Corp.
[  +0.000019] usb 1-1: SerialNumber: 00000000000583
[  +0.118897] random: nonblocking pool is initialized
[  +0.230139] ax88179_178a 1-1:1.0 eth0: register 'ax88179_178a' at usb-ff300000.usb-1, ASIX AX88179 USB 3.0 Gigabit Ethernet, 9c:eb:e8:10:5d:2d
[  +0.000098] usbcore: registered new interface driver ax88179_178a
[  +0.006590] udevd[317]: starting version 3.2.9
[  +0.008687] udevd[317]: starting eudev-3.2.9
[  +0.801759] vendor storage:20190527 ret = -1
[Sep 7 11:44] usb 1-1: USB disconnect, device number 2
[  +0.000231] ax88179_178a 1-1:1.0 eth0: unregister 'ax88179_178a' usb-ff300000.usb-1, ASIX AX88179 USB 3.0 Gigabit Ethernet
[  +0.015769] ax88179_178a 1-1:1.0 eth0 (unregistered): Failed to write reg index 0x0002: -19
[  +0.000027] ax88179_178a 1-1:1.0 eth0 (unregistered): Failed to write reg index 0x0001: -19
[  +0.000014] ax88179_178a 1-1:1.0 eth0 (unregistered): Failed to write reg index 0x0002: -19
[  +6.319837] usb 1-1: new high-speed USB device number 3 using dwc2
[  +0.190133] usb 1-1: New USB device found, idVendor=0b95, idProduct=1790
[  +0.000023] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000010] usb 1-1: Product: AX88179
[  +0.000009] usb 1-1: Manufacturer: ASIX Elec. Corp.
[  +0.000009] usb 1-1: SerialNumber: 00000000000583
[  +0.329508] ax88179_178a 1-1:1.0 eth0: register 'ax88179_178a' at usb-ff300000.usb-1, ASIX AX88179 USB 3.0 Gigabit Ethernet, 9c:eb:e8:10:5d:2d
[  +0.022397] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[  +0.321240] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[  +2.688184] ax88179_178a 1-1:1.0 eth0: ax88179 - Link status is: 1
[  +0.007211] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```
From the looks of it the device was recognized by kernel on boot but the interface was not been brought up.

usb wifi behaves very similar.

I wrote a script to print output of `nmcli d` and `ip l` in shell spring when there is no network, there is only sit0 and lo.

Remove & plug the adapter, it will work. BTW, it's still very strange though, two devices showed up, p2p0 and wlan0, only one of them would work but not the same one every time.

This was tested on the current kernel odroid-go2-kernel-4.4.189_6.

updated to odroid-go2-rrvl-kernel-4.4.189_1
===
Things becomes different, wlan0 shows up in shell spring but as "unmanaged" in `nmcli d`, after re-plug it works, and no more p2p0 device.

USB ethernet in rrvl custom kernel behaves just like that, boot as eth0 unmanaged then works after re-plug.

Jimmy-Z, 2020-09-07:
Update: I wrote another spring shell script to run `sudo nmcli dev set wlan0 managed yes`, but it doesn't work as expected :(

Jimmy-Z, 2020-09-07:
And another note, I once accidentally pressed reset button while wifi is working, then rrvl booted with wifi plugged-in and wifi works.

And I encountered another UI bug while tinkering with wifi, I was stuck in the Wi-Fi menu of RA, while pressing Y can still bring up the search input dialog, Plus(F5) + up/down brightness adjustment still works, but pressing B can't get me out, I was able to use Plus + Power to shutdown safely though, I don't know how to reproduce this bug reliably however.

Jimmy-Z, 2020-09-07:
Update: That "workaround" doesn't work with odroid-go2-kernel-4.4.189_6 since I'm getting #182 too.

mikepurvis, 2020-09-19:
I haven't got a wifi dongle which works with RRVL, but I'm seeing similar behaviour with the wired dongle I'm using in the meantime, which is [this thing](https://www.amazon.ca/UGREEN-Ethernet-1000Mbps-Compatible-Nintendo/dp/B00MYTSN18):
```
$ lsusb
Bus 001 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
```

Works after being replugged, doesn't worked if plugged in from boot.

valadaa48, 2020-12-13:
Closing this as we don't have an rgb10 to test on. This issue is still searchable for others that have similar issues.

closed 2020-12-13 by valadaa48