valadaa48/retroroller #187: Screen aspect ratio, button mapping issues on RGB10

mikepurvis, 2020-09-19:
Hey, awesome project! I'm finding it much friendlier and more configurable than the image the RGB10 came with. The process of adding cores and ROMs was very straightforward, and the scp/ssh access is great. Two quick notes below:

First, the only PSX core I've been able to get to work at all is PCSX-ReARMed r22, and even that only works with some titles (good for Soul Reaver, Castlevania SOTN, Crash Team Racing; crashes on Crash Bandicoot and hangs on Bomberman World). DuckStation doesn't work at all on any of them, just exits right away. Unfortunately, though, I'm getting screen aspect ratios all over the place on PCSX— some scenes have vertical black bars, others horizontal bars. Interestingly, the image is correctly stretched to fill the screen when the translucent "Quick Menu" is overlaid on top of it, so it's something in a runtime mode which isn't quite right.

Another small issue is that the default button mapping on this device has start and select as L2/R2, which is a bit confusing as there are actual buttons labeled start and select on the face of the device. Was this intentional, or is it just a consequence of the low-level gpio mapping being a bit different between this device and the OGA?

Thanks again and let me know if I can assist with any of this.

valadaa48, 2020-09-23:
I think you'll find that pcsxr 32bit is really the only psx emu supported on any of the images for this platform. Unsure about the issues you encountered in some games, perhaps others can chime in with their experience.

We can only support the OGA as it is the only device we have. I imagine the mappings are off due to the issues you mentioned but I can't be sure.

mikepurvis, 2020-10-27:
Yeah, my noobness is showing— I was able to remap the buttons using the built in button configurator in RetroArch and now it's much better. If there's a way to share my config so that it works out of the box for others, let me know and I'm happy to contribute.

I'll try out PCSXR, thanks for the pointer on that.

closed 2020-12-10 by valadaa48