valadaa48/retroroller #188: RRVL RA 1.9.0_17 glcore driver issue

raidenii, 2020-10-25:
Hello,

RA 1.9.0_17 has issue for glcore drivers for flycast and ppsspp core on OGA. Namely, games failed to start under RA 1.9.0_17 with video driver set to "glcore" and core option "Allow cores to switch the video driver" set to "false". In retroarch.log the following message was printed:

```
[INFO] [Environ]: GET_PREFERRED_HW_RENDER.
[INFO] [Environ]: SET_HW_RENDER.
[INFO] Requesting OpenGLES2 context.
[INFO] [Environ]: SET_HW_RENDER.
[INFO] Requesting OpenGLES2 context.
[INFO] [Environ]: SET_HW_RENDER.
[INFO] Requesting OpenGLES2 context.
[INFO] [Environ]: SET_HW_RENDER.
[INFO] Requesting OpenGLES2 context.
[ERROR] [CONTENT LOAD]: Failed to load content
```

flycast core at commit e3b8bc9 and ppsspp core at commit 6ca01d5. Downgrading retroarch to 1.9.0_16 and both cores work fine with video driver = glcore and Allow cores to switch the video driver = false.

valadaa48, 2020-10-25:
Thanks for the info. My guess is that we picked up a regression in retroarch along the way. What I will try to do is apply the retroarch work I did to the old _16 version and hopefully this issue goes away.

> flycast core at commit e3b8bc9 and ppsspp core at commit 6ca01d5. 
which commits are you referring to? Are these cores downloaded from our core updater?

raidenii, 2020-10-25:
commit of their own repo. For ppsspp it's from RRVL repo, the flycast core I compiled myself.

valadaa48, 2020-10-25:
I git bisected and indeed someone broke this in September. https://github.com/libretro/RetroArch/issues/11487

I'll see if I can revert that commit if it helps.

valadaa48, 2020-10-25:
I pushed out a new retroarch with that commit reverted. Can you try it out and close the issue if all good?

raidenii, 2020-10-25:
Unfortunately, with version 18 of RA it still failed to work.

valadaa48, 2020-10-25:
oops, hold on a sec.. forgot to sync out 20

valadaa48, 2020-10-25:
OK it's out there

raidenii, 2020-10-25:
Thanks, it works fine now.

closed 2020-10-25 by raidenii

raidenii, 2020-10-27:
Looks like it needs cores with the new code choosing renderers. Mind providing them on your repo?