valadaa48/retroroller #189: RRVL sleep/awake not work as expected on rgb10

zhangmq, 2020-10-27:
Flashed RRVL on my rgb10, found sleep/awake not work as expected, it won't go to sleep within a short time (
about a few minutes) after previous awake.  Not sure if there are other factors affecting

mikepurvis, 2020-11-12:
I have found this also, and also on an RGB10. A potentially related quirk is that I'm unable to sleep (or even screen-off) the device when it is plugged in to charge. I either need to completely power down the device, or tolerate it sitting there charging with the screen on the whole time.

valadaa48, 2020-12-13:
Closing this as we don't have an rgb10 to test on. This issue is still searchable for others that have similar issues.

closed 2020-12-13 by valadaa48